microsoft/AirSim-NeurIPS2019-Drone-Racing #3: Running AirSim headless

ricardodeazambuja, 2019-07-18:
HI,

I'm trying to run AirSim headless using VirtualGL, but I'm having some problems because I need to use the flag ``-opengl3`` and this generates some errors related to the shaders ('uncooked' stuff... I'm not used to Unreal engine). 
Error using the flag for OpenGL3:
```
[2019.07.18-02.54.35:620][  0]LogShaders: Error: The global shader cache file '/home/test/Airsim-Linux/Engine/GlobalShaderCache-GLSL_150.bin' is missing.

Your application is built to load COOKED content. No COOKED content was found; This usually means you did not cook content for this build.
It also may indicate missing cooked data for a shader platform(e.g., OpenGL under Windows): Make sure your platform's packaging settings include this Targeted RHI.

Alternatively build and run the UNCOOKED version instead.
```
... and this seems to be related to the unchecked project packaging option *Share Material Shader Code*.

Error without the flag:
```
[2019.07.18-02.57.33:109][  0]LogRHI: Error: OpenGL 4.3 not supported by driver
```
BTW, the env variables:
```
export SDL_VIDEODRIVER_VALUE=offscreen
export SDL_HINT_CUDA_DEVICE=0
```
...don't make any difference on my system.

Would you have any hints on how to run AirSim headless (or a free-of-charge way to use the Azure AirSim headless option)?

Thanks in advance!

Cheers,
Ricardo

ricardodeazambuja, 2019-07-22:
Just to clarify, headless does work using [docker](https://github.com/microsoft/AirSim/blob/master/docs/docker_ubuntu.md) on my computer (NVIDIA Driver Version: 418.56), but I can't use docker on the server I have access to (NVIDIA Driver Version: 410.79).

ricardodeazambuja, 2019-07-22:
Ok, using Singularity and the ``SDL_VIDEODRIVER_VALUE=offscreen``, it's almost working... at least it seems to be using it now:
```
[2019.07.22-19.42.41:073][  0]LogInit: Initializing SDL.
[2019.07.22-19.42.41:153][  0]LogInit: Initialized SDL 2.0.8 revision: 11184 (hg-12121:4358e537000a with EpicExtensions) (compiled against 2.0.8)
[2019.07.22-19.42.41:153][  0]LogInit: Using SDL video driver 'offscreen'
[2019.07.22-19.42.41:153][  0]LogInit: Display metrics:
[2019.07.22-19.42.41:153][  0]LogInit:   PrimaryDisplayWidth: 1024
[2019.07.22-19.42.41:153][  0]LogInit:   PrimaryDisplayHeight: 768
[2019.07.22-19.42.41:153][  0]LogInit:   PrimaryDisplayWorkAreaRect:
[2019.07.22-19.42.41:153][  0]LogInit:     Left=0, Top=0, Right=1024, Bottom=768
[2019.07.22-19.42.41:153][  0]LogInit:   VirtualDisplayRect:
[2019.07.22-19.42.41:153][  0]LogInit:     Left=0, Top=0, Right=1024, Bottom=768
[2019.07.22-19.42.41:153][  0]LogInit:   TitleSafePaddingSize: X=0.000 Y=0.000 Z=0.000 W=0.000
[2019.07.22-19.42.41:153][  0]LogInit:   ActionSafePaddingSize: X=0.000 Y=0.000 Z=0.000 W=0.000
[2019.07.22-19.42.41:153][  0]LogInit:   Number of monitors: 1
[2019.07.22-19.42.41:153][  0]LogInit:     Monitor 0
[2019.07.22-19.42.41:153][  0]LogInit:       Name: 0
[2019.07.22-19.42.41:153][  0]LogInit:       ID: display0
[2019.07.22-19.42.41:153][  0]LogInit:       NativeWidth: 1024
[2019.07.22-19.42.41:153][  0]LogInit:       NativeHeight: 768
[2019.07.22-19.42.41:153][  0]LogInit:       bIsPrimary: true
```
But it still fails:
```
[2019.07.22-19.42.42:429][  0]LogInit: Using SDL_WINDOW_OPENGL
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=52256 LargeMemoryPoolOffset=183392 
[2019.07.22-19.42.42:542][  0]LogCore: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x0000000000000010

[2019.07.22-19.42.42:542][  0]LogCore: Fatal error!
```

StanislavSychev, 2019-07-25:
I've run into the same problem and [virtual screen setup](https://gist.github.com/shehzan10/8d36c908af216573a1f0) helped me to overcome it.

madratman, 2019-07-26:
Can you look at 
https://github.com/microsoft/AirSim/blob/master/docs/docker_ubuntu.md#binaries
(ctrl+f "headless")

and the run binary linux script directly https://github.com/microsoft/AirSim/blob/master/docker/run_airsim_image_binary.sh
relevant snippets are

```
# this block is for running X apps in docker
XAUTH=/tmp/.docker.xauth
if [ ! -f $XAUTH ]
then
    xauth_list=$(xauth nlist :0 | sed -e 's/^..../ffff/')
    if [ ! -z "$xauth_list" ]
    then
        echo $xauth_list | xauth -f $XAUTH nmerge -
    else
        touch $XAUTH
    fi
    chmod a+r $XAUTH
fi
```
this thing
```
SDL_VIDEODRIVER_VALUE='';
while [ -n "$1" ]; do
    case "$1" in
    --)
        shift 
        break
        ;; 
    esac
    shift
done
 
for param in $@; do
    case "$param" in
    headless) SDL_VIDEODRIVER_VALUE='offscreen' ;;
    esac
done
```
just sets the value of the env variable to `offscreen`

madratman, 2019-07-26:
-opengl3.. I would rather recommend using the `-vulkan` flag

ricardodeazambuja, 2019-07-27:
@madratman, I'm trying to run it on Google Colaboratory so no docker, no tty, etc and that's why I tried singularity On my laptop I managed to use docker to run airsim truly headless (without passing any display or volume with X11 stuff), but on google colab... I'm still trying :)

@stanislavsych, I tried to implement the solution you suggested, but I have problems with tty (there's no on google colab). I still need to try it maybe with singularity.

madratman, 2019-10-22:
I have added support for docker. See here https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing#docker / https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/pull/100

Closing this, as we don't have the bandwidth to support google collab. 
If you figured this out, please update with your solution so others can benefit from it.

closed 2019-10-22 by madratman