microsoft/AirSim-NeurIPS2019-Drone-Racing #4: What's the proper way to capture images for the competition?

jedediyah, 2019-07-18:
Following Airsim Image API docs like the one [here](https://microsoft.github.io/AirSim/docs/image_apis/#getting-a-single-image), when I run
```
# reshape array to 4 channel image array H X W X 4
img_rgba = img1d.reshape(response.height, response.width, 4)
```
I get this error:

``` 
img_rgba = img1d.reshape(response.height, response.width, 4)  
    ValueError: cannot reshape array of size 110592 into shape (144,256,4)  
```

It seems that the dimensions being returned are actually (144,256,3), and the image is also not inverted. I can capture and view images with:
```
responses = client.simGetImages([
        airsim.ImageRequest("0", airsim.ImageType.DepthVis),  #depth visualization image
        airsim.ImageRequest("1", airsim.ImageType.DepthPerspective, True), #depth in perspective projection
        airsim.ImageRequest("1", airsim.ImageType.Scene), #scene vision image in png format
        airsim.ImageRequest("1", airsim.ImageType.Scene, False, False)])  #scene vision image in uncompressed RGB array
response = responses[3]
img1d = np.fromstring(response.image_data_uint8, dtype=np.uint8)  # get numpy array
img_rgb = img1d.reshape(response.height, response.width, 3)  # reshape array to 3 channel image array H X W X 3
cv2.imshow('RGB',img_rgb)
```
but I wonder if you could point to updated Image API docs or give an example of how best to capture images in the context of the competition.
Thanks!

ricardodeazambuja, 2019-07-18:
Hi,

I suggest you having a look at the AirSim default settings (CaptureSettings):
https://github.com/microsoft/AirSim/blob/master/docs/settings.md#available-settings-and-their-defaults

Cheers,
Ricardo

jedediyah, 2019-07-18:
Ah yes, I think I see what you're pointing to.  Probably the difference between ImageTypes with calls like:
```
airsim.ImageRequest("1", airsim.ImageType.Scene, False, False)
```
```
airsim.ImageRequest("0", airsim.ImageType.Scene, False, False)
```
But I am still curious about the particular constraints on image capture for the competition.  Particularly, what do the calls look like that will give us the images for perception and autonomy as described in the [competition site](https://www.microsoft.com/en-us/research/academic-program/game-of-drones-competition-at-neurips-2019/).

rmst, 2019-08-01:
The following improved example worked for me without warnings and errors (Python 3.7):
```python
import numpy as np
from airsimneurips import ImageRequest
import airsimneurips as airsim
client = airsim.MultirotorClient()
client.confirmConnection()
# client.simLoadLevel('Soccer_Field_Easy')  
# client.simLoadLevel('Soccer_Field_Medium')        
# client.simLoadLevel('ZhangJiaJie_Medium')
client.simLoadLevel('Building99_Hard')

response, = client.simGetImages([ImageRequest("0", airsim.ImageType.Scene, False, False)])
image = np.frombuffer(response.image_data_uint8, dtype=np.uint8).reshape(response.height, response.width, 3)  
```