microsoft/AirSim-NeurIPS2019-Drone-Racing #5: AirSim API environment navigation

anas-abulehia, 2019-07-21:
After downloading paks of the environments, i followed tutorial in github. but I am facing some problems in changing the environments. 
First one , after requesting a specific environment unreal starts loading but eventually it does not change it grey screen and drone only. 
in python shell client.simLoadLevel('Building99').
Second one , is hard levels once I request specific hard levels python shell return an error says that you enter incorrect level names as follows.
raise ValueError("Allowed level names are:  \'Building99\', \'Soccer_Field\', \'ZhangJiaJie\', or \'MainMenu\'") 

thanks in advance

yannbouteiller, 2019-07-22:
Hi, we have the same issue with environments here. The grey screen you see in the beginning is actually the default 'blocks' environment. To switch between environments, the client.simLoadLevel() command apparently doesn't work yet. You have to press F10 and change the environment manually under 'levels' (assuming you downloaded the .pak files and put them in the right folder).

anas-abulehia, 2019-07-22:
> Hi, we have the same issue with environments here. The grey screen you see in the beginning is actually the default 'blocks' environment. To switch between environments, the client.simLoadLevel() [...]

what about levels , easy , hard and etc can you change between them in F10 , I can not

jedediyah, 2019-07-22:
I also cannot switch to levels with difficulties that are not listed in the F10 menu.  I guess this functionality is not yet complete?

msb336, 2019-07-23:
The levels available in the first release are the following:
`Soccer_Field_Easy`, `Soccer_Field_Medium`, `ZhangJiaJie_Medium`, and `Building99_Hard`. These levels are also listed in the UI menu `F10`.

Are you having difficulty loading them through the API? Or only when attempting to load levels that don't exist in the release?

madratman, 2019-07-23:
can you guys do a pip upgrade package on airsimneurips. Just fixed this.

anas-abulehia, 2019-07-27:
while I was switching between different levels it seems that drone goes under the ground (does not physically  collide with ground and stopping above ground it goes below the floor ) 
it is obvious especially  when sending the following command client.simLoadLevel('Soccer_Field_Easy')

madratman, 2019-09-10:
This has been fixed in the latest binaries.

closed 2019-09-10 by madratman