microsoft/AirSim-NeurIPS2019-Drone-Racing #10: [Linux] AirSimExe only working if NOT windowed

ctoumieh, 2019-07-26:
Hi,

When running ./AirSimExe.sh in window mode, once I choose the vehicle at the start, it crashes.
This does not happen if in full screen mode. However, if I am in full screen mode and try switching workplaces or alt-tab to the terminal it also crashes.
I have a Dell XPS 9560 (GTX1050 - latest Nvidia driver and Vulkan version).
 The errors that appeared in the terminal:

[2019.07.26-18.38.32:291][  0]LogPawn: Error: EnableInput can only be specified on a Pawn for its Controller
[2019.07.26-18.38.32:557][  0]LogScript: Warning: Script Msg: Attempted to access index 0 from array 'CallFunc_GetAllActorsOfClass_OutActors' of length 0 in '/AirSim/Weather/WeatherFX/WeatherActor.WeatherActor_C:ExecuteUbergraph_WeatherActor'!
[2019.07.26-18.38.32:557][  0]LogStreaming: Error: Couldn't find file for package /AirSim/HUDAssets/MenuActor requested by async loading code. NameToLoad: /AirSim/HUDAssets/MenuActor
[2019.07.26-18.38.32:557][  0]LogStreaming: Error: Found 0 dependent packages...
[2019.07.26-18.38.32:557][  0]LogTemp: Warning: Warning, WeatherAPI got invalid menu actor class!
[2019.07.26-18.38.32:559][  0]LogTemp: PhysXCar
PhysXCar

And:

[2019.07.26-18.38.34:921][  0]LogCore: Error: appError called: Assertion failed:  [File:/home/airsim/repos/UnrealEngine/Engine/Source/Runtime/VulkanRHI/Private/VulkanMemory.cpp] [Line: 243] 
Out of Device Memory, Requested=31552.00Kb MemTypeIndex=7

[2019.07.26-18.38.34:932][  0]LogCore: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x00000000000034e7

[2019.07.26-18.38.34:932][  0]LogCore: Assertion failed:  [File:/home/airsim/repos/UnrealEngine/Engine/Source/Runtime/VulkanRHI/Private/VulkanMemory.cpp] [Line: 243] 
Out of Device Memory, Requested=31552.00Kb MemTypeIndex=7

You can find the full crash report in the zip below,

Cheers,
CT

[crash-info.zip](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/files/3437019/crash-info.zip)

ctoumieh, 2019-07-26:
Resizing the window before choosing the vehicle solved the problem.

closed 2019-07-26 by ctoumieh

Han-Sin, 2020-11-11:
> Resizing the window before choosing the vehicle solved the problem.
could you explain it more? I don't know what's the meaning.
When I run ./AirSimExe.sh -windowed -opengl4, this error appears....
Thanks a lot.

Han-Sin, 2020-11-11:
> Resizing the window before choosing the vehicle solved the problem.

really need it...please help me