microsoft/AirSim-NeurIPS2019-Drone-Racing #11: client.confirmConnection() fails

rmst, 2019-07-26:
Hi, I get the following error
```python
>>> import airsimneurips as airsim
>>> client = airsim.MultirotorClient()
>>> client.confirmConnection()
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/app/miniconda/lib/python3.7/site-packages/airsimneurips/client.py", line 54, in confirmConnection
    if self.ping():
  File "/app/miniconda/lib/python3.7/site-packages/airsimneurips/client.py", line 24, in ping
    return self.client.call('ping')
  File "/app/miniconda/lib/python3.7/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/app/miniconda/lib/python3.7/site-packages/msgpackrpc/future.py", line 43, in get
    raise self._error
msgpackrpc.error.TransportError: Retry connection over the limit
```

I'm running AirSim headless with a GPU accelerated X-server. When I take a screenshot it looks like this:
[image]

I can't interact with the UI to see what happens, though.

msb336, 2019-07-30:
It looks like you are starting your script before the airsim server has started. You can solve this by doing one of two things:

1. Edit your settings file (`Documents\Airsim\Settings.json`) to have your binary automatically load into multirotor sim:
```
{
 "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
}
```
2. Click the `No` button in that pop-up window asking whether you would like to use car simulation, and then start your airsim client script.

rmst, 2019-08-01:
Thanks! For Linux that file is `$HOME/Documents/Airsim/settings.json'. The above still fails, but it works for me with
```json
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor"
}
```
Also, for the competition, shouldn't that settings file be the default? Is there a command line option to specify the path for the settings file or do we have to trick Airsim by setting a different $HOME to prevent it from polluting the real home directory?

msb336, 2019-08-01:
For the Qualifier release and live competition the proper settings file will be included in the project executable folder, the project will use that one by default.

Airsim checks the main executable folder for a `settings.json` file before searching in `Documents/Airsim`

closed 2019-08-01 by msb336

rmst, 2019-08-02:
A perfect, thanks!