microsoft/AirSim-NeurIPS2019-Drone-Racing #7: RGB or RGBD cameras?

frontw, 2019-07-26:
Hi! 

At competition_guidelines.md, Tier II: Perception Only, two RGB cameras mentioned ("Equipped with two RGB cameras"). So, no depth information or 'D' should be added?

At Tier III: Full Autonomy there is "Given forward and downward facing RBD cameras". Do you mean RGBD cameras? Is it a typo?

Thanks!

madratman, 2019-07-29:
It's only going to be RGB cameras. We're leaning towards a single camera only, due to rendering bottlenecks with UE4

frontw, 2019-08-01:
> It's only going to be RGB cameras. We're leaning towards a single camera only, due to rendering bottlenecks with UE4

Thanks for clarify. What resolution it would have? Could we control camera orientation (by settings.json possibly).

mrpropellers, 2019-08-07:
> due to rendering bottlenecks with UE4
The scenes shown in the preview videos do not seem complex enough to cause rendering bottlenecks (what is the target fps?) You may be leveraging the SceneCaptureComponent improperly, which can force blocking calls into the render thread and break things...

madratman, 2019-09-10:
resolution is 240p Use [this script](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/baselines/generate_settings_file.py) to make a settings.json 
I think it's okay to change the orientation using these lines - https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/baselines/utils.py#L83

closed 2019-09-10 by madratman