microsoft/AirSim-NeurIPS2019-Drone-Racing #9: What API would be available to control drone?

frontw, 2019-07-26:
https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/docs/competition_guidelines.md#how-to-participate-in-the-challenge

> The simulation environment will later contain a low-level trajectory planner and feedback controller to control the dynamics of the drone to achieve the specified waypoint and velocity, thereby simplifying the interface as much as possible.

Would it be possible to control by drone velocities and attitude also?

madratman, 2019-07-26:
the `airsimneurips` pip package is a mirror of airsim pip as of now, with an addition of `simLoadLevel`, as documented in the readme. 

See the source directly to see list of supported APIs. 
https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/client.py#L279 

The link to airsim python API doc is already in the readme as well. 

yes, `moveByVelocityAsync` is available, and there is `moveByAngleThrottleAsync`

We'll be releasing two new APIs next week:
- `moveOnSplineAsync(waypoint_list, bool_add_current_position_as_constraint, bool_add_current_velocity_as_constraint, max_speed, max_acceleration)`
- `moveOnSplineVelConstraintsAsync(waypoint_list, velocity_constraints_list, bool_add_current_position_as_constraint, bool_add_current_velocity_as_constraint, max_speed, max_acceleration)`

The above will serve as high level move APIs which will handle trajectory planner and trajectory tracking control for the user.

ctoumieh, 2019-07-26:
@madratman What type of controller you will be using for the trajectory tracking?

madratman, 2019-07-26:
It's a simple pure pursuit tracking controller. chasing both position and velocity references, and tracking errors in cross-track, along-track, z direction, and yaw.
the reference yaw is obtained from velocity of the trajectory

closed 2019-08-02 by madratman

madratman, 2019-08-03:
@ctoumieh Just added APIs I was talking about in the v0.1 release. Some sample code here https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/tree/master/baselines

Just run `python generate_settings_file.py` once before you try `baseline_racer.py`.