microsoft/AirSim-NeurIPS2019-Drone-Racing #13: Maximum acceleration of drone

ctoumieh, 2019-07-29:
Hi,
While testing moveByAngleThrottle(), it appears that the maximum acceleration generated by the rotors is ~1.4g which is low for a racing drone. Are you planing on modifying it or will it be the same throughout the competition?
Cheers,
CT

yannbouteiller, 2019-07-31:
Actually I have been looking for documentation for this function (ranges of parameters values and what they mean exactly) but could not find anything. Is this information available somewhere please?

madratman, 2019-08-06:
I've updated this API, and added more lower level APIs. 
Please see https://microsoft.github.io/AirSim-NeurIPS2019-Drone-Racing/, and the recent release notes [v0.1.1](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/releases/tag/v0.1.1-linux) and [v0.1.0](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/releases/tag/v0.1.0-linux)

re making the drone more agile - we'll start working on this

closed 2019-08-30 by ctoumieh