microsoft/AirSim-NeurIPS2019-Drone-Racing #14: Tier 1 waypoints for gates.

arbaazkhan2, 2019-07-29:
For Tier 1, are the gates/levels going to be fixed with what is provided in the example levels ? 
If not is there going to be an API call that returns the pose of the next gate/all the gates ?

ironclownfish, 2019-07-29:
Hi! The practice levels that we have distributed so far won't be used for qualification or competition rounds. Different gate configurations within the same general environments will be used for those.

madratman, 2019-07-29:
@arbaazkhan2 to answer the second part of your question:

- the pip package is same as the airsim python client + simLoadLevel(), as of now. 
- You can look at the API directly here https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/client.py#L268, or in the airism docs (linked in the readme)
- Here's a little snippet to get the gate poses (this works for all levels):

```python
gate_names_sorted_bad = sorted(client.simListSceneObjects("Gate.*"))
# In building 99, for example, gate_names_sorted_bad would be ['Gate0', 'Gate10_21', 'Gate11_23', 'Gate1_3', 'Gate2_5', 'Gate3_7', 'Gate4_9', 'Gate5_11', 'Gate6_13', 'Gate7_15', 'Gate8_17', 'Gate9_19']
# number after underscore is unreal garbage. and leading zeros are missing, so the following lines fix the same
gate_indices_bad = [int(gate_name.split('_')[0][4:]) for gate_name in gate_names_sorted_bad]
gate_indices_correct = sorted(range(len(gate_indices_bad)), key=lambda k:gate_indices_bad[k])
gate_names_sorted = [gate_names_sorted_bad[gate_idx] for gate_idx in gate_indices_correct]
gate_poses = [client.simGetObjectPose(gate_name) for gate_name in gate_names_sorted]
# gate_poses has a list of gates in the order of the track now!
```

To reiterate what @ironclownfish said, yes, the current tracks are "training tracks". For quals, we'll have "test tracks". 
For planning only tier, you'd get access to the ground truth poses of gates, but that would not be the case of tiers using perception

closed 2019-07-29 by madratman