microsoft/AirSim-NeurIPS2019-Drone-Racing #18: Using AirSim Images with NumPy example fails

rmst, 2019-08-01:
I was trying out the example "Using AirSim Images with NumPy" (https://microsoft.github.io/AirSim/docs/image_apis/) but it failed. What's the problem here?

```python
Python 3.7.0 (default, Jun 28 2018, 13:15:42) 
[GCC 7.2.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import numpy as np
>>> from airsimneurips import ImageRequest
>>> import airsimneurips as airsim
>>> client = airsim.MultirotorClient()
>>> client.confirmConnection()
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

>>> # client.simLoadLevel('Soccer_Field_Easy')  
... # client.simLoadLevel('Soccer_Field_Medium')        
... # client.simLoadLevel('ZhangJiaJie_Medium')
... client.simLoadLevel('Building99_Hard')
True
>>> 
>>> responses = client.simGetImages([ImageRequest("0", airsim.ImageType.Scene, False, False)])
>>> response = responses[0]
>>> 
>>> # get numpy array
... img1d = np.fromstring(response.image_data_uint8, dtype=np.uint8) 
__main__:2: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
>>> 
>>> # reshape array to 4 channel image array H X W X 4
... img_rgba = img1d.reshape(response.height, response.width, 4)  
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
ValueError: cannot reshape array of size 110592 into shape (144,256,4)
```

frontw, 2019-08-01:
try 3 channels

rmst, 2019-08-01:
Thanks, that worked! The following improved example works without warnings and errors:
```python
import numpy as np
from airsimneurips import ImageRequest
import airsimneurips as airsim
client = airsim.MultirotorClient()
client.confirmConnection()
# client.simLoadLevel('Soccer_Field_Easy')  
# client.simLoadLevel('Soccer_Field_Medium')        
# client.simLoadLevel('ZhangJiaJie_Medium')
client.simLoadLevel('Building99_Hard')

response, = client.simGetImages([ImageRequest("0", airsim.ImageType.Scene, False, False)])
image = np.frombuffer(response.image_data_uint8, dtype=np.uint8).reshape(response.height, response.width, 3)  
```

closed 2019-08-01 by rmst