microsoft/AirSim-NeurIPS2019-Drone-Racing #19: What subset of the API are we allowed to use?

rmst, 2019-08-01:
The API offers many functions which ones will be allowed in the competition?

```python
>>> client.sim
client.simCharGetAvailableFaceExpressions(  client.simCharResetBonePose(                client.simCharSetSkinAgeing(                client.simGetGroundTruthEnvironment(        client.simGetVehiclePose(                   client.simSetObjectPose(
client.simCharGetBonePose(                  client.simCharSetBonePose(                  client.simCharSetSkinDarkness(              client.simGetGroundTruthKinematics(         client.simIsPause(                          client.simSetPose(
client.simCharGetBonePoses(                 client.simCharSetBonePoses(                 client.simContinueForTime(                  client.simGetImage(                         client.simListSceneObjects(                 client.simSetSegmentationObjectID(
client.simCharGetFaceExpression(            client.simCharSetFaceExpression(            client.simDestroyObject(                    client.simGetImages(                        client.simLoadLevel(                        client.simSetTimeOfDay(
client.simCharGetHeadRotation(              client.simCharSetFacePreset(                client.simEnableWeather(                    client.simGetObjectPose(                    client.simPause(                            client.simSetVehiclePose(
client.simCharGetSkinAgeing(                client.simCharSetFacePresets(               client.simGetCameraInfo(                    client.simGetPose(                          client.simPrintLogMessage(                  client.simSetWeatherParameter(
client.simCharGetSkinDarkness(              client.simCharSetHeadRotation(              client.simGetCollisionInfo(                 client.simGetSegmentationObjectID(          client.simSetCameraOrientation(   
```

msb336, 2019-08-01:
Stay tuned for an official release of whitelisted APIs for the competition, but as a general rule of thumb, APIs with the `sim`  prefix will be blocked. The exceptions to this being `simGetImage(s)` and `simLoadLevel`

madratman, 2019-08-02:
To add on, for control APIs, see comment here https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/9#issuecomment-515560660

madratman, 2019-08-06:
See https://microsoft.github.io/AirSim-NeurIPS2019-Drone-Racing/, and the recent release notes [v0.1.1](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/releases/tag/v0.1.1-linux) and [v0.1.0](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/releases/tag/v0.1.0-linux) for control APIs

rmst, 2019-08-07:
How does something `moveByAngleRatesThrottleAsync(roll_rate, pitch_rate, yaw_rate, throttle, duration, vehicle_name='')` work? Are the rates and throttle arguments target values that a PID controller is trying to reach within duration (by modifying the speed of the rotors) or are those set directly in the simulator and will be held for `duration` and set to zero afterwards?

Also, what happens if we call the command again before `duration` has passed? Is it replaced or will it execute _after_ the previous command finishes?

madratman, 2019-08-07:
- Throttle is set directly. There is no control for the same.    
   yes, anglerates are tried to reach with a P controller. 
   the controller runs till `duration` is reached, and then rates are set back to zero. however, wrt `throttle`, drone switches back to hover, and won't fall out of the sky

- if you call the command again before `duration` has passed, it will be replace. 
if you want it to execute `after` the previous command finishes, you can do :
`moveByAngleRatesThrottleAsync(roll_rate, pitch_rate, yaw_rate, throttle, duration, vehicle_name='').join()`
which will wait for `duration` to get finished. 

Note that for `moveByAngleRatesZAsync`, Z direction would be stabilized for you.

closed 2019-08-09 by madratman