microsoft/AirSim-NeurIPS2019-Drone-Racing #21: Different Training and Testing environments?

rmst, 2019-08-03:
Will the environments used in the final competition be the exact same as the training environments? In case the maps of the final competition environment are the same but the gate placement would be different, would there be a way to randomize gate placement during training to help with generalization?

msb336, 2019-08-05:
To answer your first question, yes the **environments** will be the same, but the **tracks** within them will not be. 

To answer your second question, there are two API calls that will help you to randomize gate positions. `simListSceneObjects` lists all of the object names in the scene (i.e. gates), and `simSetObjectPose` sets an object of a given name to your desired pose.

closed 2019-08-05 by msb336

madratman, 2019-08-06:
For completeness
 - [simListSceneObjects](https://microsoft.github.io/AirSim-NeurIPS2019-Drone-Racing/#airsimneurips.client.MultirotorClient.simListSceneObjects)
 - [simSetObjectPose](https://microsoft.github.io/AirSim-NeurIPS2019-Drone-Racing/#airsimneurips.client.MultirotorClient.simSetObjectPose)
 - https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/baselines/baseline_racer.py#L53-#L61

should pretty much tell you what to do