microsoft/AirSim-NeurIPS2019-Drone-Racing #22: baseline_racer.py problem

ctoumieh, 2019-08-06:
Hi,

I am getting the following error when running the baseline script:

```
Traceback (most recent call last):
  File "baseline_racer.py", line 190, in <module>
    main(args)
  File "baseline_racer.py", line 166, in main
    baseline_racer.initialize_drone()
  File "baseline_racer.py", line 37, in initialize_drone
    self.airsim_client.setTrajectoryTrackerGains(traj_tracker_gains.to_list(), vehicle_name=self.drone_name)
  File "/home/charbel/.local/lib/python3.6/site-packages/airsimneurips/client.py", line 224, in setTrajectoryTrackerGains
    self.client.call('setTrajectoryTrackerGains', gains, vehicle_name)
  File "/home/charbel/.local/lib/python3.6/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/charbel/.local/lib/python3.6/site-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: server could not find function 'setTrajectoryTrackerGains' with argument count 2.
```
I git pulled all repos to make sure I have the latest versions.
Any Ideas?

Cheers,
CT

ctoumieh, 2019-08-06:
My bad. Somehow the airsimneurips package wasn't upgraded. I redid it and all works fine.

closed 2019-08-06 by ctoumieh

madratman, 2019-08-06:
Thanks for trying this out, happy to take in any feedback. I just added some more lower level APIs and python version is now 0.0.11. see comment in the max acc thread
There are still some inconsistencies wrt angles and radians, but we'll clean that up soon.

FirstHandScientist, 2019-08-20:
Hi, I encounter the same problem. Could you specify how you solved it with some details?
Many thanks.
Best,
D.

ctoumieh, 2019-08-20:
Hi,
I just downloaded the latest binaries (link in readme) and did pip upgrade (instructions in the link).
CT

FirstHandScientist, 2019-08-24:
Hi, Got it. Thanks for the info.
Cheers,
D.