microsoft/AirSim-NeurIPS2019-Drone-Racing #23: enableApiControl() issue

ctoumieh, 2019-08-06:
Hi,
After I updated to version 0.0.8 everything was working fine-ish (my controller didn't work anymore with moveByRollPitchYawrateThrottleAsync() - previously worked well with moveByAngleThrottle()). Before I tried to debug my code I updated to 0.0.11. Now, with version 0.0.11 I get the following error when doing client.enableApiControl(True, vehicle_name='drone_1') - or when running the baseline script :

```
Traceback (most recent call last):
  File "baseline_racer.py", line 190, in <module>
    main(args)
  File "baseline_racer.py", line 166, in main
    baseline_racer.initialize_drone()
  File "baseline_racer.py", line 25, in initialize_drone
    self.airsim_client.enableApiControl(True, vehicle_name=self.drone_name)
TypeError: enableApiControl() got multiple values for argument 'vehicle_name'

```
Any ideas?
Cheers,
CT

frontw, 2019-08-06:
Did you update setting.json? vehicle_name is set in set here. 

```
  "Vehicles": {
    "drone_1": {
      "VehicleType": "SimpleFlight", 
...
```

madratman, 2019-08-06:
Uh, my bad. I removed the bool arg from that function and broke it into an enableApiControl and a disableApicontrol
See https://microsoft.github.io/AirSim-NeurIPS2019-Drone-Racing/#airsimneurips.client.MultirotorClient.enableApiControl
The fix is simple.. Just change that line to self.airsim_client.enableApiControl(vehicle_name=self.drone_name)

madratman, 2019-08-06:
Also, armDisarmed was broken up into an arm, and a disarm(
see, https://microsoft.github.io/AirSim-NeurIPS2019-Drone-Racing/#airsimneurips.client.MultirotorClient.arm)
The change for that line would be self.airsim_client.arm(vehicle_name=self.drone_name)

madratman, 2019-08-06:
above commit fixes this

closed 2019-08-06 by madratman

ctoumieh, 2019-08-06:
The moveByRollPitchYawrateThrottleAsync() because of the body frame change wrt world frame between the 2 versions. Now all works perfect.