microsoft/AirSim-NeurIPS2019-Drone-Racing #24: AirSim Executable Crashing Randomly After Several Baseline Racer Runs

HaniItani, 2019-08-06:
Hello, 

AirSim executable is randomly crashing after several baseline racer runs with the following error, anyone else having this issue? Is it system related? I'm running on Ubuntu18.04 with 64GB of RAM and a TitanXp 12GB VRAM GPU. Please advise.

`F0806 22:04:04.803827 18871 polynomial_optimization_linear_impl.h:297] Check failed: segment_time > 0 (-nan vs. 0) Segment times need to be greater than zero
*** Check failure stack trace: ***
Signal 6 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=6
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=118144 LargeMemoryPoolOffset=249280 
Failed to find symbol file, expected location:
"/home/${USER}/AirSimChallenge/AirSim/AirSimExe/Binaries/Linux/AirSimExe.sym"
Malloc Size=66886 LargeMemoryPoolOffset=316182 
Engine crash handling finished; re-raising signal 6 for the default handler. Good bye.
Aborted (core dumped)`

Thanks!

madratman, 2019-08-06:
Hi @Haniltani

Thanks for setting up our initial baseline. 
The error you're seeing is caused by the trajectory planner resulting in negative segment times (see `Segment times need to be greater than zero`)
As mentioned in the release notes of v0.1.1, we're building upon [mav_trajectory_generation](https://github.com/ethz-asl/mav_trajectory_generation) behind the scenes of moveOnSpline

I'll look at handling this more gracefully on the C++ side. Usually, this is caused if one sends weird position and velocity constraints to the backend. 
Can you share which specific [baseline arguments](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/baselines/baseline_racer.py#L164-#L180) were you using? I'll test this out on my side

For completeness,  the file in question is [polynomial_optimization_linear_impl.h](https://github.com/ethz-asl/mav_trajectory_generation/blob/ca0b4d4831572d601489eaea033d56d605dbeee8/mav_trajectory_generation/include/mav_trajectory_generation)/impl/polynomial_optimization_linear_impl.h)

HaniItani, 2019-08-07:
Hello @madratman,

Thank you for your reply. I am using the defaults (`all_gates_at_once`,`moveOnSpline`) so far with the default velocity/acceleration constraints. The crash does not seem to be reproducible/consistent, for example I ran the defaults on `Building99_Hard` and it crashed on the third run. Afterwards, I ran it for five consecutive times and worked flawlessly.

madratman, 2019-09-10:
This is related with https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/38 
What's happening is sometimes unreal engine returns gate poses as nans, and they are sent to moveOnSpline(), causing this line in the error log `Check failed: segment_time > 0 (-nan vs. 0)`. 
Will add an error check in next python version. 
TIll then, you can do an assert check as in the comment here https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/38#issuecomment-523267007