microsoft/AirSim-NeurIPS2019-Drone-Racing #26: Baseline  enableApiControl error

sanketgujar, 2019-08-08:
I am getting the following error while running the baseline, I have already updated the binaries and packages. 

Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

False
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Traceback (most recent call last):
  File "baseline_racer.py", line 190, in <module>
    main(args)
  File "baseline_racer.py", line 166, in main
    baseline_racer.initialize_drone()
  File "baseline_racer.py", line 25, in initialize_drone
    self.airsim_client.enableApiControl(vehicle_name=self.drone_name)
  File "/usr/local/lib/python3.5/dist-packages/airsimneurips/client.py", line 85, in enableApiControl
    return self.client.call('enableApiControl', True, vehicle_name)
  File "/usr/local/lib/python3.5/dist-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/usr/local/lib/python3.5/dist-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. The exception is not derived from std::exception. No further information available.

madratman, 2019-08-08:
Can you call python baselines/generate_settings_file.py ?

yannbouteiller, 2019-08-08:
Hi, I get the same issue, even after callin generate_settings_file.py

`
(base) yann@yann-ub:~/Desktop/GOD/Airsim/AirSim-NeurIPS2019-Drone-Racing/baselines$ python generate_settings_file.py
(base) yann@yann-ub:~/Desktop/GOD/Airsim/AirSim-NeurIPS2019-Drone-Racing/baselines$ python baseline_racer.py
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

False
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Traceback (most recent call last):
  File "baseline_racer.py", line 190, in <module>
    main(args)
  File "baseline_racer.py", line 166, in main
    baseline_racer.initialize_drone()
  File "baseline_racer.py", line 25, in initialize_drone
    self.airsim_client.enableApiControl(vehicle_name=self.drone_name)
  File "/home/yann/anaconda3/lib/python3.7/site-packages/airsimneurips/client.py", line 85, in enableApiControl
    return self.client.call('enableApiControl', True, vehicle_name)
  File "/home/yann/anaconda3/lib/python3.7/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/yann/anaconda3/lib/python3.7/site-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. The exception is not derived from std::exception. No further information available.
`

madratman, 2019-08-08:
This is a weird one, it happened only in the packaged binary of v0.1.2, but doesn't occur if I run the project in unreal engine. 
I just updated the v0.1.2-linux binaries. Please download again

closed 2019-08-08 by madratman

sanketgujar, 2019-08-08:
Thanks for the quick update :+1: