microsoft/AirSim-NeurIPS2019-Drone-Racing #28: How to get the ground truth for gates sizes?

yannbouteiller, 2019-08-10:
Hello everyone,

According to the competition guidelines, for difficulties medium and hard, the sizes of the gates will not be constant. How do we get the ground truth for these, please? I am basically simply trying to detect when the drone passes through a gate.

madratman, 2019-08-10:
See discussion [here](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/21#issuecomment-518309385) and [here](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/14#issuecomment-516199072)
and [get_ground_truth_gate_poses()](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/baselines/baseline_racer.py#L53) in baseline_racer.py
The python API is [documented here](https://microsoft.github.io/AirSim-NeurIPS2019-Drone-Racing/)

yannbouteiller, 2019-08-10:
Thank you for quick reply madratman, however I think I missed something. Don't the gate poses only contain their orientation and position ?

madratman, 2019-08-10:
ah, sorry. I misread your question. We'll get back with an answer soon.

yannbouteiller, 2019-08-10:
(By the way why I am trying to understand how AirSim uses quaternions for object orientation and the get_gate_facing_vector_from_quaternion() function in the baseline sounds quite strange to me.
I think you might have a small bug here because it returns a 4*4 identity matrix instead of a 3*1 Vector3r when n < eps.)

madratman, 2019-08-11:
Ah, yeah, that's a bug for sure, I should be returning Vector3r(0,1,0) in the edge case
Wrt how the gate vector is calculated: I am first converting the quaternion representing the gate orientation to a rotation matrix. Then, I am returning the 2nd column of the rotation matrix. The 3 columns of a rotation matrix represent the i,j,k / x,y,z basis vectors. In the unreal project, the gate mesh is set such that the gate facing vector is (0,1,0) or "j". 
I think quat * (0,1,0) * q.conj() is gonna give you the same answer.

madratman, 2019-08-23:
See latest release notes for a set and get scale API 
- https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/releases
- https://microsoft.github.io/AirSim-NeurIPS2019-Drone-Racing/#airsimneurips.client.MultirotorClient.simGetObjectScale
- https://microsoft.github.io/AirSim-NeurIPS2019-Drone-Racing/#airsimneurips.client.MultirotorClient.simSetObjectScale

madratman, 2019-08-23:
ah, we forgot to add the nominal sizes of the gates

madratman, 2019-08-29:
uh, I just noticed simSet/GetObjectScale is missing in the linux and windows binaries both, coz of some errors on our side. Will update in v0.2.1

paulerikf, 2019-08-29:
Did I understand that in Teir 3 you are allowed to get the ground truth sizes of gates?

yannbouteiller, 2019-08-29:
I don't think so, but I need that for training purpose.

msb336, 2019-08-29:
@yannbouteiller
Linux release binaries have been updated to contain the scale APIs. If you redownload [`AirSim.tar.gz`](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/releases/download/v0.2.0-linux/AirSim.tar.gz) you will be able to use said APIs.

Also, I just removed the pass-through light bulb triggers from the linux release environments and fixed some lighting issues in the B99 environment, so it might be worth redownloading the environment paks as well.

madratman, 2019-09-06:
See pic below for gate dimensions (cm in pics) and gate coord axis. (it's left-handed coz unreal engine. Red is X, Green is Y, Blue is Z)

then:  
inner dimensions (x,y,z): 0.2 m x 2.4m x 2.4 m 
outer dimensions (x,y,z):  0.2m x 3.2 m x 3.2 m

You can use the return of simGetObjectScale() and multiply that with these numbers. 
[image]
[image]
[image]

closed 2019-09-07 by madratman