microsoft/AirSim-NeurIPS2019-Drone-Racing #34: airsimneurips==0.0.14 doesn't work with corresponding AirSim-Linux.tar.gz

chisophugis, 2019-08-15:
Repro instructions:

I'm looking at: https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/releases/tag/v0.1.2

1. Download and unpack https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/releases/download/v0.1.2/AirSim-Linux.tar.gz and ZhangJiaJie.pak
2. `pip install airsimneurips==0.0.14`
3. `python baseline_racer.py`

I get the error:

```
sean:~/pg/game-of-drones/AirSim-NeurIPS2019-Drone-Racing/baselines % python baseline_racer.py
Connected!
Connected!
Traceback (most recent call last):
  File "baseline_racer.py", line 188, in <module>
    main(args)
  File "baseline_racer.py", line 164, in main
    baseline_racer.initialize_drone()
  File "baseline_racer.py", line 24, in initialize_drone
    self.airsim_client.enableApiControl(vehicle_name=self.drone_name)
  File "/home/sean/.local/lib/python2.7/site-packages/airsimneurips/client.py", line 49, in enableApiControl
    return self.client.call('enableApiControl', True, vehicle_name)
  File "/home/sean/.local/lib/python2.7/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/sean/.local/lib/python2.7/site-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. The exception is not derived from std::exception. No further information available.
zsh: exit 1     python baseline_racer.py

```

Doing `pip install airsimneurips==0.0.11` fixes the issue.

hollygrimm, 2019-08-15:
I'm running AirSim-Linux 0.1.2 and airsimneurips==0.0.14 without an error. Verify that settings.json in /home/sean/Documents/AirSim looks like this:
```{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ClockSpeed": 1,
  "Vehicles": {
    "drone_1": {
      "VehicleType": "SimpleFlight",
      "X": 0.0,
      "Y": 0.0,
      "Z": 0.0,
      "Pitch": 0.0,
      "Roll": 0.0,
      "Yaw": 0.0
    }
  }
}
```

If not, you can generate it:
```
cd baselines
python generate_settings_file.py
```

madratman, 2019-08-16:
@chisophugis, please confirm if @hollygrimm's comment solves it for you.

chisophugis, 2019-08-16:
I seem to have gotten it working. I don't think that the settings file is what fixed it though.

closed 2019-08-16 by chisophugis