microsoft/AirSim-NeurIPS2019-Drone-Racing #37: Mysterious exception thrown with `enableApiControl`

praveen-palanisamy, 2019-08-20:
**Trace**:
```bash
Traceback (most recent call last):
  File "baseline_racer.py", line 188, in <module>
    main(args)
  File "baseline_racer.py", line 164, in main
    baseline_racer.initialize_drone()
  File "baseline_racer.py", line 24, in initialize_drone
    self.airsim_client.enableApiControl(vehicle_name=self.drone_name)
  File "env/lib/python3.7/site-packages/airsimneurips/client.py", line 49, in enableApiControl
    return self.client.call('enableApiControl', True, vehicle_name)
  File "env/lib/python3.7/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "env/lib/python3.7/site-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. 
The exception is not derived from std::exception. No further information available.
```
**Env:**
- Ubuntu
- Neurips binary version: v0.1.2
- `airsimneurips` version (from PyPI): 0.0.14

frontw, 2019-08-20:
https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/23#issuecomment-518688308

closed 2019-08-20 by praveen-palanisamy

madratman, 2019-08-20:
```
$ cd baselines;
$ python generate_settings_file.py
```