microsoft/AirSim-NeurIPS2019-Drone-Racing #44: Offline trajectory generation

ctoumieh, 2019-08-28:
Hi,

I was wondering if we are allowed to generate trajectories offline (which may take an hour) well before the start of the race, especially for tier 1 where gate positions are known. Granted, overtaking maneuvers will be added real-time.

Cheers,
CT

ctoumieh, 2019-08-28:
I don't understand why not. I mean in the formula 1 (and the drone racing league I think - though not sure) they know well beforehand the starting position and map/gates.

ctoumieh, 2019-08-28:
I partially agree. First the opposing drone is not controlled so he still has to do collision/obstacle avoidance which will change said trajectory. Second, we still haven't figured out a reasonable way to get the globally time-optimal trajectory (RRT*  would give you the answer if you have infinite time) in the case of drone racing so it is still an active field of research.

ctoumieh, 2019-08-28:
I closed it by a misclick. My bad.

madratman, 2019-09-03:
yes, in tier 1 you know the gate pose ground truth. but it will be through an API so can get them just at start of race. 
I don't think you need an hour to generate trajectories. it's usually a few milliseconds with the right methods. 
re: a global time-optimal trajectory - why do you need an RRT* in the first place? The problem is that you're given a bunch of waypoints - fly through them as fast as you can (and as smooth you can (and don't hit the other drone if you can as well)). 
Time optimality is usually achieved by first assigning times to waypoints using a heuristic, and then obtaining a trajectory, and then optimizing for times (usually this step involves non-linear optimization))

in tier 2, yes you'll get noisy gate poses.

ctoumieh, 2019-09-04:
I am not using RRT*, I was just using an example. I have developed a new method that does not rely on waypoints, but on gate sizes so that it is more optimal i.e. the constraint is that the trajectory passes through the gate and not just a point. The method will determine wich optimal point at the gate it will pass through. It indeed uses non linear optimization and gives faster trajectories (that i tested) then the method you are using. (Building_99: ~13 secs, Soccer_Field_Medium:~37 secs, Soccer_Field_Easy: ~9 secs, Mountains:~38secs). It however requires a lot of time (that grows exponentially with the number of gates) to find said trajectory. It is basically an MINLP that is very hard to solve in real time, hence my question.

madratman, 2019-09-04:
hmm, for the qualifying binaries at least, you'll have enough time. 
Hmm, ok, I do see some value in releasing the track early on - I don't know much about the MINLP formulation - will look into it (googling it does throw a couple of classics https://pdfs.semanticscholar.org/8533/9ee3bd042d6118ea39316a9eb4aa9eb76854.pdf, and https://www.mit.edu/people/jhow/Richards_SM.pdf). 

I guess you have seen most of these anyway, but here's a paper for perhaps not that optimal, but real-time (grad student's interpretation of real time ofc). Might prove useful for others:
 
- C. Richter, A. Bry, and N. Roy, “Polynomial trajectory planning for aggressive quadrotor flight in dense indoor environments,” in International Journal of Robotics Research, Springer, 2016.
- M.W. Mueller, M. Hehn, and R. D'Andrea, "A computationally efficient motion primitive for quadrocopter trajectory generation," IEEE Transactions on Robotics Volume 31, no.8, pages 1294-1310, 2015.
- Helen Oleynikova, Michael Burri, Zachary Taylor, Juan Nieto, Roland Siegwart, and Enric Galceran, “Continuous-Time Trajectory Optimization for Online UAV Replanning”. In IEEE Int. Conf. on Intelligent Robots and Systems (IROS), October 2016.
- Robust and Efficient Quadrotor Trajectory Generation for Fast Autonomous Flight, Boyu Zhou, Fei Gao, Luqi Wang, Chuhao Liu and Shaojie Shen, IEEE Robotics and Automation Letters (RA-L), 2019.
- Optimal Time Allocation for Quadrotor Trajectory Generation, Fei Gao, William Wu, Jie Pan, Boyu Zhou and Shaojie Shen, IEEE/RSJ International Conference on Intelligent Robots and Systems (IROS), 2018, Madrid, Spain. full text
- Gradient-Based Online Safe Trajectory Generation for Quadrotor Flight in Complex Environments, Fei Gao, Yi Lin and Shaojie Shen
- Real-Time Planning with Multi-Fidelity Models for Agile Flights in Unknown Environments. Jesus Tordesillas, Brett T. Lopez, John Carter, John Ware, Jonathan P. How (ICRA) 2019

Thanks for sharing your results though, I'll try to up my game now :). We'll soon release a set of validation binaries and a corresponding validation leaderboard - which will help us gauge how easy (or how hard) the current tracks error.

ctoumieh, 2019-09-04:
I have actually read the papers you shared (except the last one). Thanks for sharing tho.
I'll start working on an alternative planning scheme then in case it doesn't work out.

closed 2019-09-04 by ctoumieh

ctoumieh, 2019-09-06:
@madratman I am writing a paper detailing the method I am using and I will be using airsim (and the competition maps) as a platform to compare it with other planning methods. I have a few questions:

1. Is there anything else to cite other then the airsim paper?
2. Is your trajectory generation a fair assessment of C. Richter et al.'s method for optimal time trajectory generation capabilities?

Also in **fly_through_all_gates_at_once_with_moveOnSpline** I notice you are setting **v_max** and **a_max** and passing a constant value instead of them to **airsim_client.moveOnSplineAsync**. Is this intentional?

Cheers,
CT

madratman, 2019-09-09:
@ctoumieh thanks for asking!
- Yes, I've started to write a tech report for this repo, which is in a WIP stage. Can you please share the deadline of the conference you're submitting to? I'll try to get it out before then. 
- Hmm, so this is slightly involved:
Inside the `moveOnSpline*()` calls, I am using the [non-linear optimization](https://github.com/ethz-asl/mav_trajectory_generation#nonlinear-optimization) from [ethz-asl/mav_trajectory_generation](https://github.com/ethz-asl/mav_trajectory_generation), with the following setup. 
```cpp
mav_trajectory_generation::NonlinearOptimizationParameters nlopt_parameters;
nlopt_parameters.algorithm = nlopt::LD_LBFGS;
nlopt_parameters.time_alloc_method = mav_trajectory_generation::NonlinearOptimizationParameters::kMellingerOuterLoop;
nlopt_parameters.print_debug_info_time_allocation = false;
mav_trajectory_generation::PolynomialOptimizationNonLinear<N> nlopt(D, nlopt_parameters);
```
So you, should cite the papers in their readme for sure - https://github.com/ethz-asl/mav_trajectory_generation#bibliography, and also [Mellinger&Kumar](http://www-personal.acfr.usyd.edu.au/spns/cdm/papers/Mellinger.pdf)

The key param in the above is the `time_alloc_method`, which is set to `kMellingerOuterLoop`. 
See [these lines](https://github.com/ethz-asl/mav_trajectory_generation/blob/6f941b0a842abed81e57b5274f185faa432d86de/mav_trajectory_generation/include/mav_trajectory_generation/impl/polynomial_optimization_nonlinear_impl.h#L104-#L117) on the respective internal time allocation methods - `kRichterTime` and `kMellingerOuterLoop` are different functions. 

- if `time_alloc_bound` is set to `kRichterTime`,  [`optimizeTime()`](https://github.com/ethz-asl/mav_trajectory_generation/blob/6f941b0a842abed81e57b5274f185faa432d86de/mav_trajectory_generation/include/mav_trajectory_generation/impl/polynomial_optimization_nonlinear_impl.h#L133) is called, which is setting up the objective function using [`objectiveFunctionTime()`](https://github.com/ethz-asl/mav_trajectory_generation/blob/6f941b0a842abed81e57b5274f185faa432d86de/mav_trajectory_generation/include/mav_trajectory_generation/impl/polynomial_optimization_nonlinear_impl.h#L556)

- if `time_alloc_bound` is set to `kMellingerOuterLoop`, [`optimizeTimeMellingerOuterLoop`](https://github.com/ethz-asl/mav_trajectory_generation/blob/6f941b0a842abed81e57b5274f185faa432d86de/mav_trajectory_generation/include/mav_trajectory_generation/impl/polynomial_optimization_nonlinear_impl.h#L171) is called, which is setting up the objective function using [`objectiveFunctionTimeMellingerOuterLoop()`](https://github.com/ethz-asl/mav_trajectory_generation/blob/6f941b0a842abed81e57b5274f185faa432d86de/mav_trajectory_generation/include/mav_trajectory_generation/impl/polynomial_optimization_nonlinear_impl.h#L619)

I haven't gone into the weeds of the underlying difference yet, but this should be a start. Will update after reading Richer,Bry and Mellinger, Kumar in detail again. 
http://helenol.github.io/publications/iros_2015_mav.pdf talks more about ethz's work on exploiting the sparsity of the A matrix.

ctoumieh, 2019-09-09:
I was aiming for ICRA 2020 (deadline 15 Sept. i.e. in 5 days). I'll most probably publish in it, but if not it'll be IROS 2020 (deadline in Feb 2020 which will give you enough time to finish your WIP).

So I gather that (correct me if I am wrong) this may or may not be the best possible time optimal trajectory generation moveOnSpline() is capable of? Also I am assuming there is no way I can play around with ETH's code since we are getting only the binaries?

madratman, 2019-09-09:
https://github.com/ethz-asl/mav_trajectory_generation/blob/c57ed3a7ff84ad08c4b8e3e5ddb4a3332dfe860e/mav_trajectory_generation_ros/src/time_evaluation_node.cpp

they have a benchmark script to evaluate different time allocation methods. could prove useful
For your paper, you can of course play with the ethz-asl code, as it's a standalone c++ repo. 

wrt the neurips competition, I can expose time allocation parameter via a (yet another) python argument to moveOnspline, but I think that would make the competition too easy. 
moveOnSpline() is meant to be a baseline / utility API people can use, if they want to focus on vision and not so much on planning. 
Not making it the best possible thing (in terms of time allocation and trajectory tracking controller) should fuel the competition. 

Also, yes, this may or may not be the best moveOnSpline can do in terms of time allocation and control :), and that's intentional. The best will (hopefully) be in the qualification binaries and the live competition.

madratman, 2019-09-10:
@ctoumieh, in one of their open PRs (https://github.com/ethz-asl/mav_trajectory_generation/pull/86), there's an [RTD page](https://mav-trajectory-generation.readthedocs.io/en/latest/index.html), which also [lists helpful equations for various time-alloc methods](https://mav-trajectory-generation.readthedocs.io/en/latest/pages/Time-Allocation.html). 
Might be of use.