microsoft/AirSim-NeurIPS2019-Drone-Racing #45: Methods Results

anas-abulehia, 2019-08-30:
I was trying some methods in documentation but python shell responds Object has no attribute , it seems that it does not exist ?? any solution for that guys.

Thanks in advance

madratman, 2019-08-31:
Can you please elaborate on the "some methods" part of your question. 
Please share the error msg from the terminal, along with the relevant source code snippet / line

anas-abulehia, 2019-08-31:
[image]

madratman, 2019-09-02:
please update your airsimneurips package. this API is already there 

`pip install airsimneurips --upgrade` or `pip3 install airsimneurips --upgrade`

closed 2019-09-02 by madratman