microsoft/AirSim-NeurIPS2019-Drone-Racing #48: Which drone do I control for the validation leaderboard?

ctoumieh, 2019-09-07:
Hi,

When we upload the files for the validation leaderboard, which drone does it choose to publish its results?

Cheers,
CT

msb336, 2019-09-09:
"drone_1" is the drone who's results will appear on the leaderboard. With regards to your reference frame issue, this is an expected behavior. Each drone has it's own reference frame based on your settings file, but `simGetObjectPose` always returns poses in `drone_1`'s reference frame. You will have to account for that offset on your own.

(or you can specify in your settings file that both drones spawn into the same position, and then they will share an identical reference frame)

ctoumieh, 2019-09-09:
Ok thx!

closed 2019-09-09 by ctoumieh

ctoumieh, 2019-09-09:
@msb336 I notice the leaderboard does not display french characters. I already registered with the french team name. Is that an issue or do you prefer characters without accents?

msb336, 2019-09-09:
@ctoumieh

French characters aren't an issue for us in terms of recordkeeping, but it may be some time before the leader board is able to properly display special characters

madratman, 2019-09-20:
@ctoumieh thanks for bringing this up. 
I realized they do not need to be in different reference frames as `simStartRace` just teleports the meshes. The participant can simply get the start pose (via `simGetVehiclePose` or `simGetMultirotorState`), as done in `takeoff_with_moveOnSpline()` in `baseline_racer.py`. Relevant snippet:

```python
        start_position = self.airsim_client.simGetVehiclePose(vehicle_name=self.drone_name).position
        takeoff_waypoint = airsim.Vector3r(start_position.x_val, start_position.y_val, -takeoff_height)

        self.airsim_client.moveOnSplineAsync([takeoff_waypoint], vel_max=15.0, acc_max=5.0, add_position_constraint=True, add_velocity_constraint=False, 
            add_acceleration_constraint=False, viz_traj=self.viz_traj, viz_traj_color_rgba=self.viz_traj_color_rgba, vehicle_name=self.drone_name).join()
```

ugurkanates, 2019-09-29:
Maybe kinda unrelated but  @madratman  how do i let drone_2 to move according to              fly_through_all_gates_at_once_with_moveOnSpline().join() function ? Is it bugged or am i doing something wrong

What i do is create two objects 

baseline_racer = BaselineRacer(drone_name="drone_1",viz_traj_color_rgba=[1.0, 1.0, 0.0, 1.0])
baseline_racer2 = BaselineRacer("drone_2",viz_traj_color_rgba=[1.0, 1.0, 0.0, 1.0])

Start race with either drone(doesn't matter) ,  initalise  them  and use take_off Async functions but all my commands apply for drone_1. If i dont create a drone_1 object then race starts but take off function doesn't still apply to neither drones. 
I simply couldn't way to fly drone_2 . Didnt change any settings (.json)

madratman, 2019-10-01:
@ugurkanates @ctoumieh @yannbouteiller 
we just updated the python client to version 0.3.0. 
drone_2 will now fly automatically as soon as you call `simstartrace()`

Note that for tier=2 and tier=3, simgetobjectpose() will return noisy gate poses now, so baseline_racer.py will follow these noisy poses (and probably hit stuff)
baseline_racer.py will still go through the center of gates if tier=1 is passed to `simStartRace`