microsoft/AirSim-NeurIPS2019-Drone-Racing #50: Error while trying to run baseline_racer.py

AbhinavAtrishi, 2019-09-10:
Hello, I just recently upgraded to airsimneurips v-0.20 in python. I am now getting this error while trying to run baseline_racer. Did I miss some pre-configuration? 

`Traceback (most recent call last):
  File "baseline_racer.py", line 284, in <module>
    main(args)
  File "baseline_racer.py", line 252, in main
    baseline_racer.initialize_drone()
  File "baseline_racer.py", line 63, in initialize_drone
    self.airsim_client.setTrajectoryTrackerGains(traj_tracker_gains.to_list(), vehicle_name=self.drone_name)
  File "D:\Softwares\Anaconda\envs\AirSimComp\lib\site-packages\airsimneurips\client.py", line 701, in setTrajectoryTrackerGains
    self.client.call('setTrajectoryTrackerGains', *((gains.to_list(),)+(vehicle_name,)))
AttributeError: 'list' object has no attribute 'to_list'
`

madratman, 2019-09-10:
Are you using the v0.3.0 binaries?
Airsimneurips 0.2.0 corresponds to 0.3.0 binaries.

madratman, 2019-09-10:
Ah I see you also need to git pull origin rebase from this repo
The relevant line is now https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/baselines/baseline_racer.py#L63

closed 2019-09-10 by madratman