microsoft/AirSim-NeurIPS2019-Drone-Racing #55: moveOnSplineAsync is not work

anas-abulehia, 2019-09-13:
I created my waypoints to navigate drone from a to b, I command moveOnSplineAsync(waypoints)
but drone did not move at all, please check it and let me know if this profound function is working or not ???

madratman, 2019-09-14:
Hi @anas-abulehia, the profound function christened `moveOnSpline` does work. 
Please [`baseline_racer.py`](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/baselines/baseline_racer.py) closely
Please ensure that you're calling **all** the functions in [`initialize_drone`](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/baselines/baseline_racer.py#L53) in your script before you call moveOnSpline.

madratman, 2019-09-14:
We can help debug you of course, if you can share a python snippet.

closed 2019-09-14 by madratman

anas-abulehia, 2019-09-14:
Yes , I have to define gains I define them as in initialize_drone then client.setTrajectoryTrackerGains(traj_tracker_gains), but it did not work see photo.
[image]
 Regarding values of these gains shall we tune them or it should take as it? and about maximum velocity and accelerations can we change them or we should take them as in base_line_racer.py

anas-abulehia, 2019-09-18:
[image]
Please help me in this regard