microsoft/AirSim-NeurIPS2019-Drone-Racing #56: how to run aisrim using root user

sezan92, 2019-09-15:
Hello, i figured out how to use airsim without visualization. Now i was trying to run airsim in floydhub cloud . But it uses root user. How can i run airsim despite knowing only root user?

madratman, 2019-09-15:
Is the unreal binary not running if you have a root user?
Are you using a docker container? If so, see the following dockerfile to make a user:
- https://github.com/microsoft/AirSim/blob/master/docker/run_airsim_image_binary.sh
- https://github.com/microsoft/AirSim/blob/master/docs/docker_ubuntu.md
- https://github.com/microsoft/AirSim/blob/master/docker/Dockerfile_binary

sezan92, 2019-09-26:
Hello. I am using in floydhub cloud. It uses docker container .

madratman, 2019-10-01:
I see. Does making a user as done in the dockerfile I linked above works?

madratman, 2019-10-22:
unreal binaries won't run as root user
Please see https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing#docker and you can build on the provided dockerfile - https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/docker/Dockerfile

madratman, 2019-10-22:
The key lines in the docker file are

```
RUN adduser --force-badname --disabled-password --gecos '' --shell /bin/bash airsim_user && \ 
	echo '%sudo ALL=(ALL) NOPASSWD:ALL' >> /etc/sudoers && \ 
	adduser airsim_user sudo && \ 
	adduser airsim_user audio && \ 
	adduser airsim_user video

USER airsim_user
ENV USER airsim_user
WORKDIR /home/airsim_user
RUN sudo chown -R airsim_user /home/airsim_user
```

closed 2019-10-22 by madratman