microsoft/AirSim-NeurIPS2019-Drone-Racing #57: Which sim prefix functions are OK to use in which tiers?

ugurkanates, 2019-09-15:
I got confused with reading closed issues from July to here on. On  here

https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/14

It says 

> For planning only tier, you'd get access to the ground truth poses of gates, but that would not be the case of tiers using perception

Which itself calls simListSceneObjects method/function. A prefix sim function. As far i know only sim related functions were simLoadLevel and simGetImage 

https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/19

> Stay tuned for an official release of whitelisted APIs for the competition, but as a general rule of thumb, APIs with the sim prefix will be blocked. The exceptions to this being simGetImage(s) and simLoadLevel

Can you post what is OK to use and what's not for each tier
thank you

ironclownfish, 2019-09-18:
It's possible we will allow simListSceneObjects() and simGetObjectPose() for the planning tier. As you found in that link to #14 there, that allows you to make a getGatePoses() function pretty easily. (I see your point, that those weren't listed in #19)

There's also a chance we will just supply a getGatePoses() function directly, and block off simListSceneObjects() and simGetObjectPose(). Either way, we expect people to follow the method shown in #14 for now.

yannbouteiller, 2019-09-23:
By the way, I think it is by far best for tier 3 to give access to Front and Back camera, as it was initially said.

The reason for having access to Back camera is obviously to be able to block/kill the lagging drone, which is what makes the competition adversarial and not just a "simple" path-planning task with a randomly moving obstacle.

ugurkanates, 2019-09-29:
So we are not able to get information like angular velocity,acceleration or linear velocity,acc even for our drone right ? 

baseline_racer.airsim_client.simGetGroundTruthKinematics()  - this is not allowed 

Correct me if i'm wrong. 

baseline_racer.airsim_client.simGetVehiclePose() - just this allowed for both drones

madratman, 2019-10-17:
See list for qualification binaries in the readme : https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing#downloading

as @ironclownfish mentioned, simListScneObjects is still working. 
simGetObjectPose() returns a noisy output if racer tier is passed as 2 or 3 in simStartRace

closed 2019-10-17 by madratman