microsoft/AirSim-NeurIPS2019-Drone-Racing #59: Neurips Lottery Question

ctoumieh, 2019-09-18:
Hi,
As I understand it, we have to win the lottery to be able to attend NeurIPS and compete in the live event. This means that even if one qualifies in first place, there is no garantee that he will be able to compete in the live event.
1) What are the odds of the lottery?
2) Is there a way to make competing in the live event without being physically present at the conference possible? 
Cheers,
CT

madratman, 2019-09-20:
The odds of lottery depends on the number of people who sign up NeurIPS lottery, the numbers of which we are unfortunately not aware of. 
The following snippet from NeurIPS' [official press release](https://neurips.cc/Conferences/2019/Press?article=2299) might be of use

```
Given the size of the conference center, it is estimated that 3800 tickets will go to authors, 2400 to reviewers, 1200 to presenters in workshops, 1000 to participants in co-located affinity group workshops, and 600 to conference sponsors. The conference expects that about 3500 tickets will go up for sale to the general public through the lottery.  Many reviewers and program-committee members will not register; their tickets will be added to the lottery after their 2 week invitation expires. 
```
One might expect the odds to be not that great, given the [recent surge](http://approximatelycorrect.com/2018/12/22/the-greatest-trade-show-north-of-vegas-pressing-lessons-neurips-2018/) in the field. 

For the second half of the question: In that case, a qualifying team needs to find a representative who (1) got into the reg lottery, and (2) can run your code at the conference on your behalf).

closed 2019-09-20 by madratman

ctoumieh, 2019-09-20:
I didn't realize the odds of making it to the live event were so slim when I entered the competition. I incorrectly assumed that since Microsoft was a diamond sponsor and hosting a competition, it would get a few tickets (one ticket per team representative) for the people who qualify and don't already have tickets. I totally had fun developing new algorithms for the competition and got a paper that beats the state of the art by quite a margin in time optimal trajectory generation out of it (cited your github page btw), but if I knew I wasn't (most likely) going to be at the live competition I wouldn't have put as much time into it. Hindsight 20/20 I guess ...

ashkapoor, 2019-09-20:
While NeurIPS organizers have not yet made it clear to us about extra tickets, I do think there will be some room for top performers/ presenters.

My recommendation is do enter the lottery and we if you don't get it we will try best at our end that folks who race live (top performers) can be there.

Ashish