microsoft/AirSim-NeurIPS2019-Drone-Racing #60: Random Airsim crash with segmentation fault after many iterations

yannbouteiller, 2019-09-18:
Hello,

I am testing reinforcement learning approaches in Airsim, and during training Airsim randomly crashes with a segmentation fault. I don't really understand the error, what is happening here please?

The full Airsim terminal output is available at https://pastebin.com/Mrxk3dZy

The actual error output starts at line 750 I believe. It says:

`[2019.09.18-21.47.57:359][549]LogOutputDevice: Warning:
 
Script Stack (2 frames):
CompManager_BP_C.ExecuteUbergraph_CompManager_BP
CompManager_BP_C.Reset
 
Failed to find symbol file, expected location:
"/home/yann/Desktop/GoD/AirSim/AirSimExe/Binaries/Linux/AirSimExe.sym"
Assertion failed: (Index >= 0) & (Index < ArrayNum) [File:/home/airsim/repos/UnrealEngine/Engine/Source/Runtime/Core/Public/Containers/Array.h] [Line: 611]
Array index out of bounds: 0 from an array of size 0
[2019.09.18-21.47.57:364][549]LogCore: Error: appError called: Assertion failed: (Index >= 0) & (Index < ArrayNum) [File:/home/airsim/repos/UnrealEngine/Engine/Source/Runtime/Core/Public/Containers/Array.h] [Line: 611]
 
 
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119
Malloc Size=118144 LargeMemoryPoolOffset=249280
[2019.09.18-21.47.57:375][549]LogCore: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x00000000000042e3
 
[2019.09.18-21.47.57:375][549]LogCore: Assertion failed: (Index >= 0) & (Index < ArrayNum) [File:/home/airsim/repos/UnrealEngine/Engine/Source/Runtime/Core/Public/Containers/Array.h] [Line: 611]
 
 
 
0x0000000002ecc314 AirSimExe!UnknownFunction(0x2ccc313)
0x0000000002d4e66b AirSimExe!UnknownFunction(0x2b4e66a)
0x0000000002cee6d7 AirSimExe!UnknownFunction(0x2aee6d6)
0x0000000002cee4fd AirSimExe!UnknownFunction(0x2aee4fc)
0x0000000002730f93 AirSimExe!UnknownFunction(0x2530f92)
0x0000000002730abd AirSimExe!UnknownFunction(0x2530abc)
0x00000000027324ee AirSimExe!UnknownFunction(0x25324ed)
0x0000000002fd9ed7 AirSimExe!UnknownFunction(0x2dd9ed6)
0x0000000003102333 AirSimExe!UnknownFunction(0x2f02332)
0x0000000003102dfd AirSimExe!UnknownFunction(0x2f02dfc)
0x0000000003110ef9 AirSimExe!UnknownFunction(0x2f10ef8)
0x0000000003102dfd AirSimExe!UnknownFunction(0x2f02dfc)
0x0000000002fd9ed7 AirSimExe!UnknownFunction(0x2dd9ed6)
0x000000000310557e AirSimExe!UnknownFunction(0x2f0557d)
0x0000000004a19855 AirSimExe!UnknownFunction(0x4819854)
0x0000000002702d0b AirSimExe!UnknownFunction(0x2502d0a)
0x0000000002702bc1 AirSimExe!UnknownFunction(0x2502bc0)
0x00000000026a835b AirSimExe!UnknownFunction(0x24a835a)
0x0000000002b9229d AirSimExe!UnknownFunction(0x299229c)
0x0000000002b90f74 AirSimExe!UnknownFunction(0x2990f73)
0x00000000056a2201 AirSimExe!UnknownFunction(0x54a2200)
0x000000000569d1bb AirSimExe!UnknownFunction(0x549d1ba)
0x0000000004ffd538 AirSimExe!UnknownFunction(0x4dfd537)
0x0000000004e8756f AirSimExe!UnknownFunction(0x4c8756e)
0x000000000257664b AirSimExe!UnknownFunction(0x237664a)
0x000000000257cf5a AirSimExe!UnknownFunction(0x237cf59)
0x0000000005d9995f AirSimExe!UnknownFunction(0x5b9995e)
0x00007fbd18ca4b97 libc.so.6!__libc_start_main(+0xe6)
0x000000000256c029 AirSimExe!UnknownFunction(0x236c028)
 
[2019.09.18-21.47.57:387][549]LogExit: Executing StaticShutdownAfterError
[2019.09.18-21.47.57:403][549]LogCore: Warning: Unable to statfs('/home/yann/Desktop/GoD/AirSim/AirSimExe/Saved/Crashes/crashinfo-AirSimExe-pid-17123-027EF2E008D73C60573494863D049F27/AirSimExe.log'): errno=2 (No such file or directory)
Malloc Size=75426 LargeMemoryPoolOffset=324722
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)`

yannbouteiller, 2019-09-20:
I am not sure, but I believe this issue appears when calling reset()

(I need to call reset() a lot though, because simResetRace() doesn't resets the kinematics)

madratman, 2019-09-20:
We're looking into this.
One get around to not call reset is - call simresetrace, then reset kinematics by disarming the drones, then reset the drones to be in the center of the cages by calling simresetrace one more time, and then arm the drones, and simstartrace

yannbouteiller, 2019-09-20:
Well, using this workaround might have improved the situation a little bit, but the crash still happens with it instead of reset()

yannbouteiller, 2019-09-22:
Actually, disarm() doesn't seem to reset the kinematics?

yannbouteiller, 2019-09-27:
Hello,

So, I have downloaded the latest AirSim.zip file to see if it had fixed reset(), but now whenever I call reset() and then simGetObjectPose(), simGetObjectPose() throws an error. The output I get looks like:

`
Connected!
DEBUG: begin reset
Traceback (most recent call last):
(...)
  in get_ground_truth_gate_poses_and_half_dims
    p = self.airsim_client.simGetObjectPose(object_name=gate_name)
  File "/home/yann/anaconda3/lib/python3.7/site-packages/airsimneurips/client.py", line 221, in simGetObjectPose
    pose = self.client.call('simGetObjectPose', object_name)
  File "/home/yann/anaconda3/lib/python3.7/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/yann/anaconda3/lib/python3.7/site-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'simGetObjectPose' (called with 1 arg(s)) threw an exception. The exception is not derived from std::exception. No further information available.
`

yannbouteiller, 2019-09-28:
Well, this error happens all the time even without reset(), simGetObjectPose() seems broken.

madratman, 2019-10-01:
Kinematics are now being reset with the updated binaries, when race finishes. 
The reset bug should also be fixed 
Please re-open the issue if it still persists.

closed 2019-10-01 by madratman