microsoft/AirSim-NeurIPS2019-Drone-Racing #61: Plot Transform

anas-abulehia, 2019-09-19:
Hi, sorry for opening many issues, I am trying to get some help since this is my first experience in both Python and Airsim. After defining a Pose I called function plot_transform it did not work , please call this function and tell me whether it works or not

madratman, 2019-11-11:
The API is / was working. You need to pass a list of Pose objects. Please reply if you're having issues

madratman, 2019-11-11:
Ok, I have actually found more efficient ways to implement this. In the neurips binaries, a call to plot_transform kills the game frame rate. https://github.com/microsoft/AirSim/pull/2304 offers a better implementation. I'll look into adding this.