microsoft/AirSim-NeurIPS2019-Drone-Racing #62: jupyter lab doesnt run properly with airsim

sezan92, 2019-09-20:
Hello, I wanted to run jupyter lab and /or jupyter notebook with airsim. But there are some problems with these two.. When I install airsim, i cannot run jupyter notebook properly. and also vice versa..can you tell me the proper version of jupyter for airsim so that i can run both well?

madratman, 2019-09-20:
Can you please be a bit more precise in terms of what issue you're exactly facing?
The error output along with the relevant python snippet should help. 
My hunch is the cause is the port. Can you share the output of `netstat -a` after you have started the airsim binary. 
See the [constructor of `MultirotorClient`](https://microsoft.github.io/AirSim-NeurIPS2019-Drone-Racing/api.html). the default port is 41451. 
If you need to change the port, the value has to set in `settings.json` (as shown here https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/35#issue-481027717), as well as passed through via the `MultirotorClient`'s port argument above.

sezan92, 2019-09-21:
[E 16:18:23.478 NotebookApp] Could not open static file ''
[W 16:18:23.590 NotebookApp] 404 GET /static/components/react/react-dom.production.min.js (127.0.0.1) 15.80ms referer=http://localhost:1234/tree?token=e653e89a1d483bf0d12b94360637a159c1c45454a4f9010c
[W 16:18:23.860 NotebookApp] 404 GET /static/components/react/react-dom.production.min.js (127.0.0.1) 2.25ms referer=http://localhost:1234/tree?token=e653e89a1d483bf0d12b94360637a159c1c45454a4f9010c

sezan92, 2019-09-21:
[image]

sezan92, 2019-09-21:
I got the solution. ``` pip install notebook==4.4.1```
the reason is tornado version mismatch

closed 2019-09-21 by sezan92

apertosa, 2020-09-23:
Are you able control the vehicle from the Jupyter notebook? I can connect to the server but all the subsequent cells do not get executed as if the notebook is waiting for a user input..