microsoft/AirSim-NeurIPS2019-Drone-Racing #64: How to run several instances of AirSim in parallel?

yannbouteiller, 2019-09-26:
Hello,

I am having issues on trying to launch several instances of AirSim in parallel on my machine (say, one per GPU for example) and to communicate with them through the Python API as I will eventually need to do this on a cluster to train RL algorithms. Is there any easy way of doing this, please?

Regards,
Yann.

yannbouteiller, 2019-09-26:
Sorry, I found a partial answer in this issue: https://github.com/Microsoft/AirSim/issues/563

I'll try to see if I can do something that works fine out of this.

yannbouteiller, 2019-09-26:
Reopening this issue as it seems there is a mechanism to do this in AirSim according to this issue : https://github.com/microsoft/AirSim/issues/1711 ?

I don't understand how we can change the port of the launched server when executing
./AirSimExe.sh -windowed

Can you help, please ?

madratman, 2019-09-26:
Server port is set in settings.json, and that file is automatically read by the unreal binary when you start it via `./AirSimExe.sh -windowed`. See this comment for another example : https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/35#issue-481027717

Client port can be passed as a param in the constructor of multirotorclient https://microsoft.github.io/AirSim-NeurIPS2019-Drone-Racing/api.html#airsimneurips.client.MultirotorClient 

The workaround you mentioned is saying set port to a value - say 41451 - and run one instance, then modify the settings.json to a new port value - say 41452 - and run another instance. 
now, you can pass either 41451/41452 to multirotorclient to talk to either instance of airsim.

yannbouteiller, 2019-09-26:
Well, it works when modifying the LocalHostIp value, for example setting
"LocalHostIp": "127.0.0.1"
for the first run and
"LocalHostIp": "127.0.0.2"
for the second run, but it crashes when trying to do
"ApiServerPort": 41451
and
"ApiServerPort": 41452
instead. I am doing something wrong?

madratman, 2019-09-26:
Here's another workaround. 
AirSim gives higher priority to a `settings.json` lying in the _same_ directory as the linux/windows unreal binary, as compared to the one lying in `/home/$USER/Documents/AirSim/settings.json`

So, one way to do this is to copy paste the unreal binaries, and have different settings.json in each binary folder. 
 
For example, you can place a `settings.json` in 
`PATH_TO/AirSim_1/AirSimExe/Binaries/Linux/settings.json`, with a particular port number, and when you run it with `./AirSim_1/AirSimExe.sh -windowed`

And then you can copy-paste the unreal engine binaries:
`PATH_TO/AirSim_2/AirSimExe/Binaries/Linux/settings.json`, with a new port number, and when you run it with `./AirSim_2/AirSimExe.sh -windowed` it should use the new port number

I know either workaround is suboptimal. We'll look into passing the path to a json file as a command line arg.

madratman, 2019-09-26:
yeah, it's crashing with new port with multiple instances :/. Looking into it.

madratman, 2019-09-27:
I just updated the linux v0.3.0's AirSim.zip  (pak files are the same)
It shouldn't crash now with different ports. A

madratman, 2019-09-27:
The new zip also has a fix for the reset bug from your other thread. Let us know if it's fixed now.

yannbouteiller, 2019-09-27:
Well, the port issue seems fixed, thank you.

However, about reset(), I get an error when calling GetObjectPose() after calling reset() apparently, I will report it in the other thread.

closed 2019-09-27 by yannbouteiller

madratman, 2019-09-28:
Ah, I ll update the pythonclient.

yannbouteiller, 2019-09-29:
Hi @madratman 

Did you update it, please? Because nothing happens when I try to pip-install airsimneurips --upgrade and I still get the crash.

madratman, 2019-10-02:
This should be updated now