microsoft/AirSim-NeurIPS2019-Drone-Racing #70: new simStartRace function( python client 0.3) gives exception

ugurkanates, 2019-10-02:
rpclib: function 'simGetObjectPose' (called with 2 arg(s)) threw an exception

 >>> baseline_racer.start_race(1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/paypaytr/Desktop/MicrosoftSimulation/AirSim/theCodeGiven/baselines/baseline_racer.py", line 46, in start_race
    self.airsim_client.simStartRace(tier)
  File "/home/paypaytr/anaconda3/lib/python3.7/site-packages/airsimneurips/client.py", line 140, in simStartRace
    competitor.gate_poses_ground_truth = [self.__internalRandomGoalZone(gate) for gate in sorted(self.simListSceneObjects(".*[Gg]ate.*"))]
  File "/home/paypaytr/anaconda3/lib/python3.7/site-packages/airsimneurips/client.py", line 140, in <listcomp>
    competitor.gate_poses_ground_truth = [self.__internalRandomGoalZone(gate) for gate in sorted(self.simListSceneObjects(".*[Gg]ate.*"))]
  File "/home/paypaytr/anaconda3/lib/python3.7/site-packages/airsimneurips/client.py", line 167, in __internalRandomGoalZone
    gate_pose = self.__internalGetObjectPose(gate_name)
  File "/home/paypaytr/anaconda3/lib/python3.7/site-packages/airsimneurips/client.py", line 280, in __internalGetObjectPose
    pose_msgpack = self.client.call('simGetObjectPose', object_name, False)
  File "/home/paypaytr/anaconda3/lib/python3.7/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/paypaytr/anaconda3/lib/python3.7/site-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'simGetObjectPose' (called with 2 arg(s)) threw an exception. The exception is not derived from std::exception. No further information available.

Ubuntu 18.04 , Linux , Same on all levels and tiers

madratman, 2019-10-02:
Hey can you update your Python client - should be 0.3.0 -  and also re download airsim.zip? (no need to re-download pak files) 
Sorry for the trouble, but we added noisy poses for tier 2 and 3, hence the change in signature.

ugurkanates, 2019-10-02:
Oh okay its fixed now

closed 2019-10-02 by ugurkanates