microsoft/AirSim-NeurIPS2019-Drone-Racing #71: Problem with simStartRace() and enableApiControl('drone_2') in the new binary

yannbouteiller, 2019-10-03:
Hello,

I am encountering a huge problem with the new airsimneurips binary. It seems that it has been modified so only drone_1 can be controlled while drone_2 does its stuff.

This is fine for evaluating against the reference algorithm I guess, but this is a disqualifying issue for me because I need to be able to control both drones to apply my Reinforcement Learning strategies, and I need the simulator and race to reset as fast as possible (it seems that the taking off of drone_2 adds a huge time overhead before I can control drone_1).

How can I disable the baseline controller of drone_2 and use my own controller for both drones again, please?

At the moment, not only calling enableApiControl('drone_2') makes subsequent call to simStartRace() block the program, but drone_2 will want to do its stuff anyway.

Regards,
Yann.

madratman, 2019-10-03:
Hey. Here's what you can do to disable `drone_2` from flying around the scene and take control over it:

- Figure out the path to the directory holding the airsimneurips python package on your machine:
    `$ python3.6 -c "import airsimneurips, os; print(os.path.dirname(os.path.abspath(airsimneurips.__file__)))"`

This returns the following in my machine, for example:
    `/usr/local/lib/python3.6/dist-packages/airsimneurips`

- Now, open up `/usr/local/lib/python3.6/dist-packages/airsimneurips/client.py` with root permissions, if needed. (Or I guess, if you're using a virtualenv / docker container, it's better), and find the following lines, defining the `simStartRace` function

    ```python
         # Race APIs
        def simStartRace(self, tier=1):
            """ Starts an instance of a race in your given level, if valid."""
            self.race_tier = tier
            competitor = BaselineRacer(self, viz_traj=False, viz_traj_color_rgba=[1.0, 1.0, 0.0, 1.0])
            competitor.level_name = self.level_name
            competitor.initialize_drone()
            competitor.gate_poses_ground_truth = [self.__internalRandomGoalZone(gate) for gate in sorted(self.simListSceneObjects(".*[Gg]ate.*"))]
            if self.level_name == 'Soccer_Field_Medium':
                curr_pose = deepcopy(competitor.gate_poses_ground_truth[19])
                curr_pose.position.x_val = (competitor.gate_poses_ground_truth[19].position.x_val + competitor.gate_poses_ground_truth[20].position.x_val) / 2
                competitor.gate_poses_ground_truth.insert(20, curr_pose)

            if self.level_name == 'Building99_Hard':
                competitor.gate_poses_ground_truth.insert(3, Pose(Vector3r((-21.49-41.89)/2.0, (-5.44-1.84)/2, (1.51+1.82)/2)))

            if self.level_name == 'ZhangJiaJie_Medium':
                num_gates = len(competitor.gate_poses_ground_truth)
                last_gate_position = deepcopy(competitor.gate_poses_ground_truth[-1].position)
                second_last_gate_position = deepcopy(competitor.gate_poses_ground_truth[-2].position)
                competitor.gate_poses_ground_truth.insert(num_gates-1, Pose(Vector3r((last_gate_position.x_val+second_last_gate_position.x_val)/2.0, 
                                                                                (last_gate_position.y_val+second_last_gate_position.y_val)/2.0,
                                                                                (last_gate_position.z_val+second_last_gate_position.z_val)/2.0 )))

            takeoff_height = 1.0
            start_position = self.simGetVehiclePose("drone_2").position
            takeoff_waypoint = Vector3r(start_position.x_val, start_position.y_val, start_position.z_val-takeoff_height)
            self.client.call('simStartRace', tier)
            # competitor.gate_poses_ground_truth.insert(0, Pose(takeoff_waypoint))    
            competitor.takeoff_with_moveOnSpline(takeoff_height)
            competitor.fly_through_all_gates_at_once_with_moveOnSpline()
    ```

    Now, you can modify the above to the following, thereby disable `drone_2`:

    ```python
         # Race APIs
        def simStartRace(self, tier=1):
            """ Starts an instance of a race in your given level, if valid."""
            self.race_tier = tier
            self.client.call('simStartRace', tier)
    ```

madratman, 2019-10-03:
Hmm, you're correct in that `drone_2` is blocking everything while it's taking off. We'll look into it:
This is happening coz of the line `competitor.fly_through_all_gates_at_once_with_moveOnSpline()`, which is called with a `.join()` in `baseline.py`. 
Maybe we should use another thread for controlling `drone_2`

yannbouteiller, 2019-10-03:
The manual fix for controlling drone_2 seems to be working, thanks @madratman :)

madratman, 2019-10-03:
Awesome! Closing then

closed 2019-10-03 by madratman

madratman, 2019-10-16:
Fixed the blocking call to takeoff for drone_2 in pythonclient. Please pip install --upgrade airsimeneurips to 1.0.0