microsoft/AirSim-NeurIPS2019-Drone-Racing #73: baseline_racer.py throws error

Ujwal2910, 2019-10-07:
running the baseline_racr.py throws the following error-

Traceback (most recent call last):
  File "baseline_racer.py", line 278, in <module>
    main(args)
  File "baseline_racer.py", line 244, in main
    baseline_racer = BaselineRacer(drone_name="drone_1", viz_traj=args.viz_traj, viz_traj_color_rgba=[1.0, 1.0, 0.0, 1.0], viz_image_cv2=args.viz_image_cv2)
  File "baseline_racer.py", line 19, in __init__
    self.airsim_client = airsim.MultirotorClient()
  File "/home/ujwal/anaconda3/lib/python3.7/site-packages/airsimneurips/client.py", line 26, in __init__
    self.client = msgpackrpc.Client(msgpackrpc.Address(ip, port), timeout = timeout_value, pack_encoding = 'utf-8', unpack_encoding = 'utf-8')
  File "/home/ujwal/anaconda3/lib/python3.7/site-packages/msgpackrpc/client.py", line 15, in __init__
    loop.attach_periodic_callback(self.step_timeout, 1000) # each 1s
  File "/home/ujwal/anaconda3/lib/python3.7/site-packages/msgpackrpc/loop.py", line 39, in attach_periodic_callback
    self._periodic_callback.start()
  File "/home/ujwal/anaconda3/lib/python3.7/site-packages/tornado/ioloop.py", line 887, in start
    self._schedule_next()
  File "/home/ujwal/anaconda3/lib/python3.7/site-packages/tornado/ioloop.py", line 915, in _schedule_next
    self._update_next(self.io_loop.time())
  File "/home/ujwal/anaconda3/lib/python3.7/site-packages/tornado/ioloop.py", line 922, in _update_next
    callback_time_sec *= 1 + (self.jitter * (random.random() - 0.5))
TypeError: unsupported operand type(s) for *: 'AsyncIOLoop' and 'float'

madratman, 2019-10-07:
Please pip uninstall all versions of airsimneurips on your machine. (run pip uninstall airsimneurips multiple times until nothing is left). And then pip install airsimneurips again.

Ujwal2910, 2019-10-09:
Yeah thanks. Actually I needed to install rpc-msgpack as an admin and run airsim at the same time to get the connection established.

closed 2019-10-09 by Ujwal2910