microsoft/AirSim-NeurIPS2019-Drone-Racing #76: Airsim as an Openai Gym env

Ujwal2910, 2019-10-09:
I have been successful in establishing a connection with aimsun by running the baseline_racer after running airsim. After this, the drone flies in some trajectory and then crashes or possibly wanders. I wanted to sample the action space as in any gym environment or manually supply actions for the drone for testing purposes.

How can I integrate the following functionality in the code-

```
import gym
env = gym.make('CartPole-v0') # possibly start the communication via the client here for Airsim
env.reset()
for _ in range(1000):
    env.step(env.action_space.sample()) # take a random action
env.close() 
```
I want to manually supply action so that once I know how the actions are being, the type and properties of the action can I replace the random sampling of action space with a Refined piece of Reinforcement learning snippet or model. Help would be appreciated.

yannbouteiller, 2019-10-09:
I would recommend using the API to build your reward function, and then follow these instructions: https://github.com/openai/gym/blob/master/docs/creating-environments.md

Beware though, generally this will only work for tier 2.

ugurkanates, 2019-10-12:
It's not that easy situtation as copy and paste. You need to create your own state,reward function, done etc.. I suggest learning more on creating custom enviroments in general maybe with ML unity agents

closed 2019-10-14 by Ujwal2910

Deepakgthomas, 2021-01-31:
@yannbouteiller Why do you say that the integration of Airsim and OpenAI gym will work with tier 2. Could you please elaborate?

Thanks!

yannbouteiller, 2021-02-01:
> @yannbouteiller Why do you say that the integration of Airsim and OpenAI gym will work with tier 2. Could you please elaborate?
> 
> Thanks!

Well it actually works on all tiers, but tiers 1 and 3 are multi-agent so it comes with additional difficulties.

I have not announced it because I work on other things these days, the project in on hold and not in a presentable state, but I have silently open sourced this : https://github.com/yannbouteiller/gym-airsimdroneracinglab, where basically I have implemented a OpenAI Gym environment for multi-agent tiers as well. You may find it interesting.

Deepakgthomas, 2021-02-01:
Sounds good. Thank you.