microsoft/AirSim-NeurIPS2019-Drone-Racing #77: AirSim crashes on reset() when one drone has finished the track

yannbouteiller, 2019-10-09:
Hello,

Once again, reset() seems responsible for AirSim crashes. Now it makes the simulator crash each time it is called after either drone has finished the track. The output I get from the simulator is:

> Signal 11 caught.
> Malloc Size=65538 LargeMemoryPoolOffset=65554 
> CommonUnixCrashHandler: Signal=11
> Malloc Size=65535 LargeMemoryPoolOffset=131119 
> Malloc Size=118144 LargeMemoryPoolOffset=249280 
> Failed to find symbol file, expected location:
> "/home/yann/Desktop/GoD/Algorithm/environments/gym-airsimneurips/gym_airsimneurips/envs/AirSim/AirSimExe/Binaries/Linux/AirSimExe.sym"
> [2019.10.09-19.57.22:148][227]LogCore: === Critical error: ===
> Unhandled Exception: SIGSEGV: unaligned memory access (SIMD vectors?)
> 
> [2019.10.09-19.57.22:149][227]LogCore: Fatal error!
> 
> 0x0000000002747307 AirSimExe!UnknownFunction(0x2547307)
> 0x000000000274702b AirSimExe!UnknownFunction(0x254702a)
> 0x000000000274e20d AirSimExe!UnknownFunction(0x254e20c)
> 0x0000000002fe9057 AirSimExe!UnknownFunction(0x2de9056)
> 0x00000000031146fe AirSimExe!UnknownFunction(0x2f146fd)
> 0x00000000027112fb AirSimExe!UnknownFunction(0x25112fa)
> 0x00000000027111b1 AirSimExe!UnknownFunction(0x25111b0)
> 0x0000000002747c77 AirSimExe!UnknownFunction(0x2547c76)
> 0x000000000274e626 AirSimExe!UnknownFunction(0x254e625)
> 0x0000000002fe9057 AirSimExe!UnknownFunction(0x2de9056)
> 0x00000000031114b3 AirSimExe!UnknownFunction(0x2f114b2)
> 0x000000000311830f AirSimExe!UnknownFunction(0x2f1830e)
> 0x0000000003111f7d AirSimExe!UnknownFunction(0x2f11f7c)
> 0x0000000003120079 AirSimExe!UnknownFunction(0x2f20078)
> 0x0000000003111f7d AirSimExe!UnknownFunction(0x2f11f7c)
> 0x0000000003120079 AirSimExe!UnknownFunction(0x2f20078)
> 0x0000000003111f7d AirSimExe!UnknownFunction(0x2f11f7c)
> 0x0000000002fe9057 AirSimExe!UnknownFunction(0x2de9056)
> 0x00000000031146fe AirSimExe!UnknownFunction(0x2f146fd)
> 0x0000000004a289b5 AirSimExe!UnknownFunction(0x48289b4)
> 0x00000000027112fb AirSimExe!UnknownFunction(0x25112fa)
> 0x00000000027111b1 AirSimExe!UnknownFunction(0x25111b0)
> 0x0000000004cd4237 AirSimExe!UnknownFunction(0x4ad4236)
> 0x0000000004cd8244 AirSimExe!UnknownFunction(0x4ad8243)
> 0x0000000004cd16b3 AirSimExe!UnknownFunction(0x4ad16b2)
> 0x0000000004d0f51c AirSimExe!UnknownFunction(0x4b0f51b)
> 0x0000000004a3dc39 AirSimExe!UnknownFunction(0x483dc38)
> 0x000000000267c4b7 AirSimExe!UnknownFunction(0x247c4b6)
> 0x000000000269c48d AirSimExe!UnknownFunction(0x249c48c)
> 0x00000000025e2c37 AirSimExe!UnknownFunction(0x23e2c36)
> 0x00000000026d29e8 AirSimExe!UnknownFunction(0x24d29e7)
> 0x00000000025e2c37 AirSimExe!UnknownFunction(0x23e2c36)
> 0x00000000026a1bcc AirSimExe!UnknownFunction(0x24a1bcb)
> 0x0000000002671abc AirSimExe!UnknownFunction(0x2471abb)
> 0x000000000268fd19 AirSimExe!UnknownFunction(0x248fd18)
> 0x00000000026a3074 AirSimExe!UnknownFunction(0x24a3073)
> 0x00000000026b641b AirSimExe!UnknownFunction(0x24b641a)
> 0x0000000002ba145d AirSimExe!UnknownFunction(0x29a145c)
> 0x0000000002ba0134 AirSimExe!UnknownFunction(0x29a0133)
> 0x000000000567b577 AirSimExe!UnknownFunction(0x547b576)
> 0x00000000025860ce AirSimExe!UnknownFunction(0x23860cd)
> 0x0000000002589f5a AirSimExe!UnknownFunction(0x2389f59)
> 0x0000000005da8b7f AirSimExe!UnknownFunction(0x5ba8b7e)
> 0x00007fccd90f6b97 libc.so.6!__libc_start_main(+0xe6)
> 0x0000000002579029 AirSimExe!UnknownFunction(0x2379028)
> 
> [2019.10.09-19.57.22:162][227]LogExit: Executing StaticShutdownAfterError
> [2019.10.09-19.57.22:173][227]LogCore: Warning: Unable to statfs('/home/yann/Desktop/GoD/Algorithm/environments/gym-airsimneurips/gym_airsimneurips/envs/AirSim/AirSimExe/Saved/Crashes/crashinfo-AirSimExe-pid-17229-027E948E08D74CD15EFB852CC903AF59/AirSimExe.log'): errno=2 (No such file or directory)
> Malloc Size=69240 LargeMemoryPoolOffset=318536 
> Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
> Segmentation fault (core dumped)

After NeurIPS please tell Microsoft they should hire me as a beta tester haha

yannbouteiller, 2019-10-11:
Hello, sorry but this bug is very annoying in my case (it makes everything crash as soon as aither drone crosses the last gate (by mere luck) during my training experiments). Is it possible to provide a fix for this, please?

By the way I am using the modded version of the API to control both drones.

ironclownfish, 2019-10-15:
We will look into this. Can you attach the log file produced during the crash?
(in Saved/Logs)

yannbouteiller, 2019-10-15:
[15-10-2019_18-57-49_level_Soccer_Field_Easy_tier_2_1.log](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/files/3731755/15-10-2019_18-57-49_level_Soccer_Field_Easy_tier_2_1.log)

[crashinfo-AirSimExe-pid-23897-02C6C89708D751A1A4E645F29487C934.zip](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/files/3731758/crashinfo-AirSimExe-pid-23897-02C6C89708D751A1A4E645F29487C934.zip)

Hello @ironclownfish , attached are the logs and the crashinfo of an example, but it is very easy to reproduce: it happens every time we call reset() after crossing the last gate

yannbouteiller, 2019-10-20:
Hello, is there progress made on this issue, please ? We must really get rid of simulator crashes and freezes to be able to train something.

madratman, 2019-10-21:
@yannbouteiller  we just updated the binaries with a fix to these reset bugs. can you please re-download airsim.zip and let us know if it works now?

madratman, 2019-10-21:
fwiw, I added a little reproducible test script we used to verify reset(), simResetRace() are robust to calling from multiple threads here https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/tests/test_reset.py

yannbouteiller, 2019-10-22:
Hello @madratman , sorry but using the training binaries the bug still happens. Only the error output has changed a little bit, it is now:

> 
> Signal 11 caught.
> Malloc Size=65538 LargeMemoryPoolOffset=65554 
> CommonUnixCrashHandler: Signal=11
> Malloc Size=65535 LargeMemoryPoolOffset=131119 
> Malloc Size=118144 LargeMemoryPoolOffset=249280 
> Failed to find symbol file, expected location:
> "/home/yann/Desktop/GoD/Algorithm/environments/gym-airsimneurips/gym_airsimneurips/envs/AirSim/AirSimExe/Binaries/Linux/AirSimExe.sym"
> [2019.10.22-20.24.34:515][514]LogCore: === Critical error: ===
> Unhandled Exception: SIGSEGV: unaligned memory access (SIMD vectors?)
> 
> [2019.10.22-20.24.34:515][514]LogCore: Fatal error!
> 
> 0x0000000002748467 AirSimExe!UnknownFunction(0x2548467)
> 0x0000000002748346 AirSimExe!UnknownFunction(0x2548345)
> 0x000000000274f1bd AirSimExe!UnknownFunction(0x254f1bc)
> 0x0000000002fe9fc7 AirSimExe!UnknownFunction(0x2de9fc6)
> 0x000000000311568e AirSimExe!UnknownFunction(0x2f1568d)
> 0x00000000027122fb AirSimExe!UnknownFunction(0x25122fa)
> 0x00000000027121b1 AirSimExe!UnknownFunction(0x25121b0)
> 0x0000000002748be7 AirSimExe!UnknownFunction(0x2548be6)
> 0x000000000274f5d6 AirSimExe!UnknownFunction(0x254f5d5)
> 0x0000000002fe9fc7 AirSimExe!UnknownFunction(0x2de9fc6)
> 0x0000000003112443 AirSimExe!UnknownFunction(0x2f12442)
> 0x000000000311929f AirSimExe!UnknownFunction(0x2f1929e)
> 0x0000000003112f0d AirSimExe!UnknownFunction(0x2f12f0c)
> 0x0000000003121009 AirSimExe!UnknownFunction(0x2f21008)
> 0x0000000003112f0d AirSimExe!UnknownFunction(0x2f12f0c)
> 0x0000000003121009 AirSimExe!UnknownFunction(0x2f21008)
> 0x0000000003112f0d AirSimExe!UnknownFunction(0x2f12f0c)
> 0x0000000002fe9fc7 AirSimExe!UnknownFunction(0x2de9fc6)
> 0x000000000311568e AirSimExe!UnknownFunction(0x2f1568d)
> 0x0000000004a29905 AirSimExe!UnknownFunction(0x4829904)
> 0x00000000027122fb AirSimExe!UnknownFunction(0x25122fa)
> 0x00000000027121b1 AirSimExe!UnknownFunction(0x25121b0)
> 0x0000000004cd5167 AirSimExe!UnknownFunction(0x4ad5166)
> 0x0000000004cd9174 AirSimExe!UnknownFunction(0x4ad9173)
> 0x0000000004cd25e3 AirSimExe!UnknownFunction(0x4ad25e2)
> 0x0000000004d1044c AirSimExe!UnknownFunction(0x4b1044b)
> 0x0000000004a3eb89 AirSimExe!UnknownFunction(0x483eb88)
> 0x000000000267d4b7 AirSimExe!UnknownFunction(0x247d4b6)
> 0x000000000269d48d AirSimExe!UnknownFunction(0x249d48c)
> 0x00000000025e3c37 AirSimExe!UnknownFunction(0x23e3c36)
> 0x00000000026d39e8 AirSimExe!UnknownFunction(0x24d39e7)
> 0x00000000025e3c37 AirSimExe!UnknownFunction(0x23e3c36)
> 0x00000000026a2bcc AirSimExe!UnknownFunction(0x24a2bcb)
> 0x0000000002672abc AirSimExe!UnknownFunction(0x2472abb)
> 0x0000000002690d19 AirSimExe!UnknownFunction(0x2490d18)
> 0x00000000026a4074 AirSimExe!UnknownFunction(0x24a4073)
> 0x00000000026b741b AirSimExe!UnknownFunction(0x24b741a)
> 0x0000000002ba248d AirSimExe!UnknownFunction(0x29a248c)
> 0x0000000002ba10d3 AirSimExe!UnknownFunction(0x29a10d2)
> 0x0000000002b9f515 AirSimExe!UnknownFunction(0x299f514)
> 0x00000000056b2631 AirSimExe!UnknownFunction(0x54b2630)
> 0x00000000056ad26b AirSimExe!UnknownFunction(0x54ad26a)
> 0x000000000500cf60 AirSimExe!UnknownFunction(0x4e0cf5f)
> 0x0000000004e9760f AirSimExe!UnknownFunction(0x4c9760e)
> 0x000000000258464b AirSimExe!UnknownFunction(0x238464a)
> 0x000000000258af5a AirSimExe!UnknownFunction(0x238af59)
> 0x0000000005da994f AirSimExe!UnknownFunction(0x5ba994e)
> 0x00007f6087266b97 libc.so.6!__libc_start_main(+0xe6)
> 0x000000000257a029 AirSimExe!UnknownFunction(0x237a028)
> 
> [2019.10.22-20.24.34:528][514]LogExit: Executing StaticShutdownAfterError
> [2019.10.22-20.24.34:544][514]LogCore: Warning: Unable to statfs('/home/yann/Desktop/GoD/Algorithm/environments/gym-airsimneurips/gym_airsimneurips/envs/AirSim/AirSimExe/Saved/Crashes/crashinfo-AirSimExe-pid-15117-028FC32508D7570C5351AC3E6FC20D27/AirSimExe.log'): errno=2 (No such file or directory)
> Malloc Size=70950 LargeMemoryPoolOffset=320246 
> Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
> Segmentation fault (core dumped)

yannbouteiller, 2019-10-22:
PS: I had not thought about it, but maybe this happens when we use reset() while the simulator is paused. I will give it further testing later tonight.

madratman, 2019-10-22:
@yannbouteiller yeah, we haven't tested this with sim paused. 
Can you please give us a reproducible script, something on the lines of https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/tests/test_reset.py?

cc @ironclownfish

yannbouteiller, 2019-10-23:
Ok, the bug happens with and without the simulator being paused while calling reset(). I don't understand how you don't have it because it happens all the time when one drone has crossed the last gate on my machine. Are the Linux training binaries really up to date ? I have downloaded them from https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/releases/tag/v0.3.0-linux again and I still get the crash.

Attached is a script that I used to test. It requires sudo because of the keyboard module. Use the numpad to make drone_1 cross the last gate so you see 'drone_1 has finished', and press q to call reset.

This makes airsim crash almost every time on my machine (Ubuntu 18.04.2 LTS)

[test_keyboardcontroller.zip](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/files/3760048/test_keyboardcontroller.zip)

yannbouteiller, 2019-10-23:
(No I am running one single process and one single thread here, when I want to parallelize I use one single-threaded process for each AirSim instance, so calls to the API are serialized to avoid issues with concurrent calls)

ironclownfish, 2019-10-23:
Thanks for the repro steps & script -- it proved extremely helpful.
I believe the problem is clear. I'll update once it's confirmed and fixed.

ironclownfish, 2019-10-23:
We have implemented a fix and uploaded new binaries. We were able to crash every time with your repro steps, now the crash is gone. Please re-download and give it a try.

yannbouteiller, 2019-10-24:
Well I have just re-downoaded the linux training binaries, tried again, and it crashed at first try :/

I have built a workaround that litteraly consists in deleting the last gate.

madratman, 2019-10-24:
@yannbouteiller I think we had a human error in updating the linux binaries. 
I just updated them yet again. 
here's the test script I use - https://github.com/madratman/AirSim-NeurIPS2019-Drone-Racing/blob/tests/one_drone_end_race_and_reset_called/baselines/repro_crash_one_drone_ends_race_and_reset_called.py 
I have turned off drone_2's motion by changing `airsimneurips/baseline.py` and ported your `dummy_reset()` in the above script 

Also, you need to unpause the sim - https://github.com/madratman/AirSim-NeurIPS2019-Drone-Racing/blob/tests/one_drone_end_race_and_reset_called/baselines/repro_crash_one_drone_ends_race_and_reset_called.py#L13 - before calling resetrace, or else the drones are not in the starting cage. 

Thanks for your patience in enduring with this. 

```python
    def takeoff_and_fly_through_all_gates_at_once_with_moveOnSpline(self):
        self.takeoff_with_moveOnSpline(1.0)
        # self.fly_through_all_gates_at_once_with_moveOnSpline()
```

yannbouteiller, 2019-10-24:
After some testing, it seems that this bug is indeed gone. Thank you!

closed 2019-10-24 by yannbouteiller

madratman, 2019-10-24:
Yay, thanks @yannbouteiller and @ironclownfish

ugurkanates, 2019-10-27:
It still happens to me sadly. I will provided detailed logs,my hardware setup and video to reproduce & show problem