microsoft/AirSim-NeurIPS2019-Drone-Racing #78: How to kill a specific instance of the simulator?

yannbouteiller, 2019-10-09:
Hello,

Now I am running several instances of the simulator in parallel smoothly, but the problem arises when I try to kill a specific instance of AirSim. To launch an instance I do something like:

> 
> EXECUTABLE_NAME = 'AirSimExe.sh'
> OPTIONS = '-windowed'
> dir_path = os.path.join(dir_path, 'AirSim', EXECUTABLE_NAME)
> p = subprocess.Popen([dir_path, OPTIONS])

But later calls to either

> p.terminate()

or

> p.kill()

simply try to kill the wrong process, ie not a PID that corresponds to the command

> .../AirSim/AirSimExe/Binaries/Linux/AirSimExe AirSimExe -windowed

How to programmatically terminate a specific instance of the simulator, please?

Regards,
Yann

madratman, 2019-10-10:
This - https://stackoverflow.com/questions/4789837/how-to-terminate-a-python-subprocess-launched-with-shell-true - seems relevant. 

See https://stackoverflow.com/questions/7989922/opening-a-process-with-popen-and-getting-the-pid and https://stackoverflow.com/questions/31039972/python-subprocess-popen-pid-return-the-pid-of-the-parent-script

I think you need to pass `shell=True`. Can you try it? 

I am not sure how useful the following would be (probably unclean to mix shell and python):

If I run the binary via terminal, and grep for the processes there are two
```
$ ps aux | grep AirSim
madratm+ 16918  0.0  0.0   4508   752 pts/30   S+   09:40   0:00 /bin/sh ./AirSimExe.sh -windowed
madratm+ 16925  183  1.7 3078344 1121064 pts/30 Sl+ 09:40   0:51 /home/madratman/Documents/Binaries/Neurips/v0.3.0-port/AirSim/AirSimExe/Binaries/Linux/AirSimExe AirSimExe -windowed
madratm+ 17103  0.0  0.0  21292  1072 pts/31   S+   09:41   0:00 grep --color=auto AirSim
```

To kill the binary, I need `16925`: 
`$ kill -9 16925`,
which can automated by a: 
`kill -9 $(ps aux | grep /Linux/AirSimExe | awk '{print $2}')`

yannbouteiller, 2019-10-11:
Ok, it is pretty extreme but their solution seems to work if we replace SIGTERM by SIGKILL.

So:

> p = subprocess.Popen(dir_path + ' ' + OPTIONS, stdout=subprocess.PIPE, shell=True, preexec_fn=os.setsid)

and

> os.killpg(os.getpgid(p.pid), signal.SIGKILL)

closed 2019-10-11 by yannbouteiller