microsoft/AirSim-NeurIPS2019-Drone-Racing #79: Get nearby Gate coordinates

Ujwal2910, 2019-10-10:
Is there an API function to get gate coordinates in the order they have to be passed?
Possibly I want to know that I have passed gate 1 and hence my next coordinate should be gate 2''s.
Help would be appreciated

madratman, 2019-10-14:
How about checking for L2 distance? How can use the odometry_callback defined in the baseline racer script to achieve this

Here's a quick snippet:

```python
	# define appropriate variables in the baseline racer class, something like:
	# self.next_gate_idx = 0 
	# self.last_gate_passed_idx = -1 
	# self.gate_passed_thresh = self.gate_diagonal_length / 2
	# self.gate_passed_thresh = 0.5 # or 1.0 if lazy 

    def odometry_callback(self):
        self.curr_position_linear = self.airsim_client_odom.getMultirotorState().kinematics_estimated.position

        dist_from_next_gate = math.sqrt( (self.curr_position_linear.x_val - self.gate_poses_ground_truth[self.next_gate_idx].position.x_val)**2
                                        + (self.curr_position_linear.y_val - self.gate_poses_ground_truth[self.next_gate_idx].position.y_val)**2
                                        + (self.curr_position_linear.z_val- self.gate_poses_ground_truth[self.next_gate_idx].position.z_val)**2)

        if dist_from_next_gate < self.gate_passed_thresh:
            self.last_gate_passed_idx += 1
            self.next_gate_idx += 1
```

Ujwal2910, 2019-10-15:
Oh, this is very useful. This can help structure me the reward function also. Thanks a ton @madratman

madratman, 2019-10-15:
No problem. Glad it helped.

closed 2019-10-15 by madratman

madratman, 2019-10-16:
You can also build on the log_monitor.py script to handle gate passed. 
https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/scripts/logging/log_monitor.py#L51

Here's a little snippet from a log file:
```
drone_2 time 17621 gates_passed 2
drone_1 time 17646 odometry_XYZRPY (5532.530,3097.350,-1060.876,0.022,-0.014,86.657)
drone_2 time 17646 odometry_XYZRPY (5743.078,3784.857,-1049.107,-0.513,0.909,88.947)
drone_2 time 17646 gates_passed 2
drone_1 time 17675 odometry_XYZRPY (5533.286,3110.519,-1063.196,0.000,0.000,86.654)
drone_2 time 17675 odometry_XYZRPY (5743.477,3798.413,-1047.432,-0.494,0.942,88.916)
drone_2 time 17675 gates_passed 2
drone_1 time 17705 odometry_XYZRPY (5534.041,3123.677,-1065.532,-0.059,0.016,86.651)
drone_2 time 17705 odometry_XYZRPY (5743.883,3811.943,-1045.770,-0.476,0.975,88.884)
drone_2 time 17705 gates_passed 2
drone_1 time 17735 odometry_XYZRPY (5534.796,3136.823,-1067.882,-0.138,0.037,86.647)
drone_2 time 17735 odometry_XYZRPY (5744.297,3825.446,-1044.122,-0.458,1.007,88.851)
drone_2 time 17735 gates_passed 2
drone_1 time 17759 odometry_XYZRPY (5535.400,3147.332,-1069.773,-0.201,0.058,86.644)
drone_2 time 17759 odometry_XYZRPY (5744.631,3836.228,-1042.815,-0.444,1.033,88.825)
drone_2 time 17759 gates_passed 2
drone_1 time 17784 odometry_XYZRPY (5536.006,3157.833,-1071.673,-0.257,0.081,86.641)
drone_2 time 17784 odometry_XYZRPY (5744.971,3846.993,-1041.517,-0.431,1.059,88.799)
```

you can run the log_monitor.py in a separate python thread and keep track of which drone passed which gate

surfii3z, 2019-11-09:
So we can use the log information e.g. "drone_1" has passed gate 1 in the qualification tier2?