microsoft/AirSim-NeurIPS2019-Drone-Racing #81: Does enemy drone take off and move faster than our drone?

ugurkanates, 2019-10-13:
After v0.3 it feels like my drone can't react and fast as predetermined-coded drone. When start_race() function is called enemy drone immediately does takeOff while my function call of takingOff is next line. 

`baseline_racer.start_race(3)
baseline_racer.takeoffAsync()
`
My drone is lagging on this because our drone doesn't take off with start_race function. Also after calling takeoffAsync() it feels like its waiting for sometime to adjust itself after its completed its move. So while other drone is started to race I'm still unable to move/command.

`baseline_racer.airsim_client.moveByVelocityAsync(1, 1, 1,1).join()
`
Which is the function what I call lets say for this example.

I created a gif to demonstrate my problem. (Sorry for low resolution)
[image]

madratman, 2019-10-16:
Fixed the blocking call to takeoff for drone_2 in pythonclient. Please pip install --upgrade airsimeneurips to 1.0.0

madratman, 2019-10-16:
Also, in general, takeoffasync is slow to finish. calling takeoffasync().join() is a blocking call
you can try the takeoff_with_moveonspline() function as well.

closed 2019-10-16 by madratman