microsoft/AirSim-NeurIPS2019-Drone-Racing #86: Camera parameters

collinabidi, 2019-10-16:
I'm trying to find camera calibration and distortion parameters (aside from the resolution and FOV) so I can plug recorded RGB-D video into ORB_SLAM2

saihv, 2019-10-16:
You can approximate the camera intrinsics through the following formula, where HFOV is 90 degrees by default. 

Horizontal FoV = 2 * arctan( width / 2f ) (estimate f through this, assuming f_x = f_y = f
c_x = w/2 ; c_y = h/2

The drone camera also has only very negligible distortion - assuming the coefficients as zero should work fairly well.

madratman, 2019-10-16:
You can transcribe this C++ snippet from our ROS wrapper on the main AirSim repo to python - https://github.com/microsoft/AirSim/blob/master/ros/src/airsim_ros_pkgs/src/airsim_ros_wrapper.cpp#L1043-#L1061. This essentially implements the equation @saihv mentions above

To add on, you can see the distortion params are in the order of 0.000x here https://github.com/microsoft/AirSim/issues/269#issuecomment-313526623 -

closed 2019-10-31 by madratman