microsoft/AirSim-NeurIPS2019-Drone-Racing #92: Segmentation ID's

rachthree, 2019-10-17:
Hello,

Is there a way to get segmentation ID's using the API when an image is taken? Right now, I get RGB when using `simGetImages([airsim.ImageRequest(camera_name, airsim.ImageType.Segmentation, pixels_as_float=False, compress=False)])` according to the mapping (https://github.com/microsoft/AirSim/blob/master/docs/seg_rgbs.txt, which I think is off... something I know is segmentation ID 0 shows as `[130, 219, 128]` instead of `[55, 181, 57]`). I can think of a workaround, but if there is an API call to know the segmentation ID off the bat when an image is taken (especially if it's an alpha channel), that would be very helpful.

Thanks!

madratman, 2019-10-21:
use this as your ~/Documents/AirSim/settings.json

```json
{
  "ClockSpeed": 1,
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
    "drone_1": {
      "Cameras": {
        "fpv_cam": {
          "CaptureSettings": [
            {
              "FOV_Degrees": 90,
              "Height": 240,
              "ImageType": 0,
              "Width": 320
            },
            {
              "Height": 240,
              "ImageType": 5,
              "Width": 320
            }
          ],
          "Pitch": 0.0,
          "Roll": 0.0,
          "X": 0.25,
          "Y": 0.0,
          "Yaw": 0.0,
          "Z": 0.0
        }
      },
      "Pitch": 0.0,
      "Roll": 0.0,
      "VehicleType": "SimpleFlight",
      "X": 0.0,
      "Y": 0.0,
      "Yaw": 0.0,
      "Z": 0.0
    },
    "drone_2": {
      "Pitch": 0.0,
      "Roll": 0.0,
      "VehicleType": "SimpleFlight",
      "X": 0.0,
      "Y": 0.0,
      "Yaw": 0.0,
      "Z": 0.0
    }
  }
}
```

And you get the rgb and segmentation images as:

```python
requests = [airsim.ImageRequest("fpv_cam", airsim.ImageType.Scene, False, False), airsim.ImageRequest("fpv_cam", airsim.ImageType.Segmentation, False, False)]
responses = self.airsim_client_images.simGetImages(request)
# responses[0] will be rgb
# responses[1] will be segmentation 
```

re: segmentation ids - that file is old. 
You can do set the segmentation of ID of all object except gates to be the same number by using [simSetSegmentationObjectID](https://microsoft.github.io/AirSim-NeurIPS2019-Drone-Racing/api.html#airsimneurips.client.MultirotorClient.simSetSegmentationObjectID), then you can do a numpy.unique on the segmentation image to figure out the gate rgb value, as at the end of the day, you need a binary segmentation mask

rachthree, 2019-10-21:
Thank you for the suggestion! That is very similar to what I had to do, so I’m glad to hear I was going in the right direction. I ended up pointing the computer vision camera at the ground, changed the ground segmentation ID for all ID’s, and mapped the resulting segmentation RGB to the ID.  It would be nice for AirSim to provide such masks already as this would greatly help utilize it for perception with deep learning methods. I will probably create an issue there after the competition.

madratman, 2019-10-21:
glad to know it helped, @rachthree. 
I'll add some sample code to the repo to generate a dataset.

closed 2019-11-11 by madratman