microsoft/AirSim-NeurIPS2019-Drone-Racing #95: Log Error: Nice try. But this function is not supported in the qualification binaries

ugurkanates, 2019-10-17:
Log gives me this error , but it doesn't specify which function its supposed to mean

madratman, 2019-10-17:
it's not exactly an error: the functions just return some garbage result, and the unreal binary log on terminal spits out the "nice try..." message. 
not that these APIs still work in the previous release / v0.3.0 training binaries.

ugurkanates, 2019-10-18:
I don't use these 
"simSetObjectPose
simSetVehiclePose
simSetObjectScale
simGetObjectScale
simSetSegmentationObjectID
simGetSegmentationObjectID
simPause
simContinueForTime
" Functions in my code , although I use baseline_racer as library in my code wonder if it uses it?

ugurkanates, 2019-10-19:
Failure. No convergence after 20 shifts. Program terminated.

It also started to giving these

madratman, 2019-10-21:
Can you re-downloading the linux binaries? I am not able to reproduce the error. 

"Failure. No convergence after 20 shifts. Program terminated." -> this seems like an error message from your code.

madratman, 2019-10-21:
hmm, I was wrong - apologies!
some of the banned functions are being called by simstartrace() in the backend. 
However, the log stops spitting out a couple of seconds after simstartrace() has called - this doesn't really have an effect on you. You can ignore those spurious logs when simstartrace is called, it's a red herring. 
I'll try getting rid of them. 

"Failure. No convergence after 20 shifts. Program terminated." - this seems to be your code.

madratman, 2019-10-24:
"Failure. No convergence after 20 shifts. Program terminated." - this seems to be your code.

I was wrong about the above. This error message is thrown by one of our functions in the trajectory planning backend we're using - a  polynomial root finder at this line https://github.com/ethz-asl/mav_trajectory_generation/blob/c57ed3a7ff84ad08c4b8e3e5ddb4a3332dfe860e/mav_trajectory_generation/src/rpoly/rpoly_ak1.cpp#L373

This is occuring very rarely it seems.

madratman, 2019-10-24:
@ugurkanates thanks for pointing this out. 
I just fixed these issues in v1.0 binaries. You need to 
- re-download the qualification binaries. use `./download_qualification_binaries.sh`
- upgrade airsimneurips api to 1.1.0 `$ pip install airsimneurips --upgrade`

closed 2019-10-24 by madratman

ugurkanates, 2019-10-27:
Thanks for update @madratman , i was sick so couldn't write you back.