microsoft/AirSim-NeurIPS2019-Drone-Racing #97: Need to use the banned API functions for training purpose!

yannbouteiller, 2019-10-22:
Good morning everyone,

So, I have downloaded the new binaries in hope crashes were fixed and I could launch training sessions smoothly, but I had a bad surprise : I am using RL approaches and my training environment heavily relies on newly 'not supported' API functions for training purpose. For example, I need to get the ground truth of everything to compute my reward function, and I need to use simPause() to use on-policy optimizers. These banned API functions don't need to be used in the final controller of course, but they definitely need to be used during training.

How can we use them to train our controller in the new binaries, please?

(PS: I cannot even try whether the reset() bug has been fixed now because my manual testing script works with my RL training environment.)

yannbouteiller, 2019-10-22:
(PPS: I can use tier 1 to train everything, but even this doesn't seem to be working)

yannbouteiller, 2019-10-22:
(closing this since I may have misunderstood)

closed 2019-10-22 by yannbouteiller

madratman, 2019-10-22:
@yannbouteiller 
I just updated the v0.3 binaries as well. Apologies, only updated v1.0 yesterday. 
Can you give it a shot now?
see https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/94#issuecomment-545080883

madratman, 2019-10-22:
yeah, we have disabled these APIs only in the qualification binaries, you have them all in the training binaries. 
As the environments are the same, it shouldn't affect the training - barring the reset bug which is (hopefully) fixed now.

madratman, 2019-10-22:
To test the reset bug fix, you can peruse https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/tests/test_reset.py 

Change [L79](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/tests/test_reset.py#L79) level name for training / qualification levels.