microsoft/AirSim-NeurIPS2019-Drone-Racing #105: Gates not fully rendering

ctoumieh, 2019-10-27:
Hi,

This is not really an issue/bug, but for tier 2 and 3 I am using the checkers pattern on the gate to detect it. It is however not rendering when the drone is far from the gate approx. more then 1 meters (probably due to my low quality GPU - GTX 1050 mobile and LoD used in the engine), so when I take the picture the gate is white and no pattern appears.

Is there a way to resolve this?

Cheers,
CT

ugurkanates, 2019-10-28:
I would suggest lowering graphical settings maybe ( AirSim/saved/gameSettings.ini) something like this. But that might also  effect a little bit of your performance on Vision related tasks.

ctoumieh, 2019-10-28:
Thx for your suggestion. I tried modifying `AirSim_Qualification/AirSimExe/Saved/Config/LinuxNoEditor/GameUserSettings.ini` but it didn't solve the issue.

madratman, 2019-10-29:
`./AirsimExe.sh. -windowed -opengl4` 

https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/README.md#running

ctoumieh, 2019-10-29:
Yep. That solved it. My bad, didn't recently check the readme.

closed 2019-10-29 by ctoumieh