microsoft/AirSim-NeurIPS2019-Drone-Racing #106: rpclib : server couldn't find function simGetObjectScaleInternal

ugurkanates, 2019-10-27:
Hey first of all huge thanks for docker setups you provided and bash scripts they are hugely useful.
The error is I have set up docker image and it's working on cloud PC. 
Waiting for any client code connection (like logging stuff etc no error or anything like that)

Then I connect my client code which I use on local PC. client objects connect and it prints Connected etc (odom,image) . Then it tries to run
baseline_racer.start_race(1) ( I use baseline_racer.py as import , on my code) This is current version of baseline_racer (with a little log  for debug)

```
Traceback (most recent call last):
  File "metest.py", line 449, in <module>
    baseline_racer.start_race(1)
  File "/home/ugurkanates97/theCode/NeurIRS2019DroneChallengeRL/src/source/baseline_racer.py", line 56, in start_ra
ce
    self.airsim_client.simStartRace(tier)
  File "/home/ugurkanates97/.local/lib/python3.7/site-packages/airsimneurips/client.py", line 145, in simStartRace
    competitor.gate_poses_ground_truth = [self.__internalRandomGoalZone(gate) for gate in sorted(self.simListSceneO
bjects(".*[Gg]ate.*"))]
  File "/home/ugurkanates97/.local/lib/python3.7/site-packages/airsimneurips/client.py", line 145, in <listcomp>
    competitor.gate_poses_ground_truth = [self.__internalRandomGoalZone(gate) for gate in sorted(self.simListSceneO
bjects(".*[Gg]ate.*"))]
  File "/home/ugurkanates97/.local/lib/python3.7/site-packages/airsimneurips/client.py", line 172, in __internalRan
domGoalZone
    gate_scale = self.simGetObjectScaleInternal(gate_name)
  File "/home/ugurkanates97/.local/lib/python3.7/site-packages/airsimneurips/client.py", line 343, in simGetObjectS
caleInternal
    scale = self.client.call('simGetObjectScaleInternal', object_name)
  File "/home/ugurkanates97/.local/lib/python3.7/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/ugurkanates97/.local/lib/python3.7/site-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: server could not find function 'simGetObjectScaleInternal' with argument count 1
.
```

This is the crash output on my code.  Docker supposedly (looking from files) pulls all new content and training files etc. I have no problem with that.

```
Name: airsimneurips
Version: 1.1.0
Summary: Python package for Game of Drones - A NeurIPS 2019 Competition, built on Microsoft AirSim - an open source
 simulator based on Unreal Engine for autonomous vehicles from Microsoft AI & Research
Home-page: https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing
Author: Ratnesh Madaan, Matthew Brown, Nicholas Gyde, Sai Vemprala, Shital Shah
Author-email: ratnesh.madaan@microsoft.com, v-mattbr@microsoft.com, v-nigyde@microsoft.com, sai.vemprala@microsoft.
com, shitals@microsoft.com
License: MIT
Location: /home/ugurkanates97/.local/lib/python3.7/site-packages
Requires: numpy, msgpack-rpc-python
Required-by: 
```
This is pip show of installed airsimneurips in cloud PC. It looks alright?

I'm not sure what seems to be the problem.

ugurkanates, 2019-10-27:
This only happens with tier 1 and tier 3 btw. If you call simStartRace(2) it works somehow ?

madratman, 2019-10-30:
Thanks for bringing this up, fixed the training binaries. Please re-download

closed 2019-10-30 by madratman

Sohojoe, 2019-10-31:
I have the same error running on Windows - I only downloaded today using https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/releases/tag/v1.0-windows

madratman, 2019-10-31:
@Sohojoe sorry, didn't update the windows binaries (it started to look like we had zero windows users). @ironclownfish is on it now. He'll ping when he's done.

MicroYY, 2019-11-01:
I have the same problem and I am trying to build the lasest source code.

ironclownfish, 2019-11-01:
@Sohojoe 
@MicroYY 
Go ahead and download the new binaries (AirSim.zip) and try again.