microsoft/AirSim-NeurIPS2019-Drone-Racing #109: What is the allowed way to get images?

yannbouteiller, 2019-11-01:
Hello,

I would like to know what is the allowed way to get images 2 and 3, please (I mean the exact API call). Right now, I am doing this (I have modified the .json file so drone_2 also has a fpv_cam with the same settings as drone_1):

> im1 = airsim_client.simGetImage(camera_name="fpv_cam", image_type=0, vehicle_name='drone_1')
> im2 = airsim_client.simGetImage(camera_name="fpv_cam", image_type=0, vehicle_name='drone_2')

yannbouteiller, 2019-11-01:
This doesn't work by the way. I get the same image for both vehicles. How can I get them for each drone?

madratman, 2019-11-01:
unfortunately, airsim doesn't allow same camera names. use fpv_cam_2, and this should work. 
using either simgetimage or simgetimages is fine. 
we don't expect people to be looking at what drone_2 is seeing. we expect them to get the pose of drone_2 directly

yannbouteiller, 2019-11-01:
Yes, images from drone_2 are only for training purpose in my case. I'll try this, thanks

madratman, 2019-11-01:
I think simGetImages is faster than simGetImage. You can test with this script https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/baselines/baseline_racer_image_benchmarker.py

yannbouteiller, 2019-11-02:
Oh yes now I see why you were preoccupied about passing more than one image per drone. Even passing 1 image per drone with simGetImages is extremely slow on my laptop, maybe 4 FPS. What about using a much lower resolution?

yannbouteiller, 2019-11-03:
> unfortunately, airsim doesn't allow same camera names. use fpv_cam_2, and this should work.
> using either simgetimage or simgetimages is fine.
> we don't expect people to be looking at what drone_2 is seeing. we expect them to get the pose of drone_2 directly

This solved this question by the way, so I close this issue, I will be opening another one for simGetImages() huge overhead.

closed 2019-11-03 by yannbouteiller