microsoft/AirSim-NeurIPS2019-Drone-Racing #112: How to run the simulator headless ?

yannbouteiller, 2019-11-05:
Hello, I have been looking at these issues:
https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/3
https://github.com/Microsoft/AirSim/issues/1224
But these didn't help. I need to use the simulator without the rendering window (and ideally without any rendering at all for tier 1), without using the docker or anything that requires sudo. How can we do this, please?

madratman, 2019-11-06:
https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/docker/run_docker_image.sh#L62 setting this env variable manually doesn't do it?

yannbouteiller, 2019-11-06:
No it doesn't seem to have any effect, but setting DISPLAY='' instead seems to work in the sense that it doesn't pop a window.

This doesn't disable the rendering for tier 1 though.

madratman, 2019-11-06:
yes, that is an option as well. `$ DISPLAY= ./AirSimExe.sh -opengl4` is the oneliner you need. 
To completely kill everything for tier 1, you can use `./AirSimExe.sh -nullrhi` - not that if you call simGetImages with `-nullrhi` option, it would result in an error.

yannbouteiller, 2019-11-07:
Ok thanks !

closed 2019-11-07 by yannbouteiller