microsoft/AirSim-NeurIPS2019-Drone-Racing #114: AirSimExe raises SIGABRT on incorrect vertices size

layman-n-ish, 2019-11-07:
AirSimExe raises signal 6 (SIGABRT) on an assertion fail:
```F1106 23:44:11.933295 31485 vertex.cpp:257] Check failed: vertices.size() >= 2 (1 vs. 2) ```

This signal is raised well into the execution of my code i.e. after ~3-4 hrs of training and I fail to recognise the source. Any help would be appreciated. 

Please find attached the complete log and crash files. 
[GoD_crash.log](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/files/3818691/GoD_crash.log)
[Diagnostics.txt](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/files/3818694/Diagnostics.txt)

The client side ceases with: 

``` Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/nishant/Desktop/GoD/baselines/baseline_racer.py", line 215, in repeat_timer_odometry_callback
    task()
  File "/home/nishant/Desktop/GoD/airsim_gym/envs/AirSim_Client.py", line 44, in odometry_callback
    drone_state = self.airsim_client_odom.getMultirotorState(vehicle_name=self.drone_name).kinematics_estimated
  File "/home/nishant/.local/lib/python3.6/site-packages/airsimneurips/client.py", line 1053, in getMultirotorState
    return MultirotorState.from_msgpack(self.client.call('getMultirotorState', vehicle_name))
  File "/home/nishant/.local/lib/python3.6/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/nishant/.local/lib/python3.6/site-packages/msgpackrpc/future.py", line 43, in get
    raise self._error
msgpackrpc.error.TimeoutError: Request timed out 
```
Just to confirm, the timeout raised here is due to the server side being terminated, right?

madratman, 2019-11-07:
Are you using moveOnSpline? If you are passing an empty vecot

layman-n-ish, 2019-11-08:
I use moveOnSpline() only during takeoff. Thereafter, I use moveByVelocity().

The termination has occurred only 2-3 times now; I sense the behaviour is very erratic. Will still try to reproduce it and upload it here.

madratman, 2019-11-10:
Hmm, try putting an assert to whatever you are sending to moveOnSpline and make sure the waypoint makes sense. Also, see comment here https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/116#issuecomment-551945947

madratman, 2019-11-11:
@layman-n-ish any news? If you can give us a reproducible script, it would really help in nailing this down, or else the cause should be similar to my comment in issue 116 (linked above)

layman-n-ish, 2019-11-12:
Sorry for the late reply, was away. 

Are you sure this assertion fail is from moveOnSpline()? Because, as I said, I use moveOnSpline() only in the takeoff API - takeoff_with_moveOnSpline() (that has the ```add_position_constraint``` set as True) and I use moveByVelocityAsync() thereafter. 

takeoff_with_moveOnSpline() is called on reset which is similar to the ```dummy_reset()``` function you provided earlier:
```
def _reset(self):
		self.airsim_client.simPause(True)
		self.airsim_client.reset()
		# time.sleep(0.5)
		self.airsim_client.enableApiControl(vehicle_name=self.drone_name)
		self.airsim_client.arm(vehicle_name=self.drone_name)
		self.airsim_client.simPause(False) 
		self.airsim_client.simResetRace()
		self.airsim_client.simStartRace(tier=self.race_tier)
		self.takeoff_with_moveOnSpline() 
```

On the otherhand, I'll work on the reproducible script.

madratman, 2019-11-13:
@layman-n-ish see my comment here https://gist.github.com/madratman/e617b53ec20c5f38a7d10633ba3a42c9/revisions. This could be the source of your problem as well. 
You don't need simPause  and unpause I think

madratman, 2019-11-13:
@layman-n-ish see my last 3 comments in this thread https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/123 . It should fix your issue as well. Please try using the reset function I added in the gist here https://gist.github.com/madratman/e617b53ec20c5f38a7d10633ba3a42c9#file-airsim_neurips_reset_episode_crash_repro-py-L43

```python
        self.airsim_client.reset()
        self.airsim_client.simResetRace()
        time.sleep(1.0)
        self.airsim_client.simStartRace(tier=1)
```

layman-n-ish, 2019-11-14:
Thank you for the solution. I am trying it out now. Seems promising till now. Will let you know how it goes.

layman-n-ish, 2019-11-14:
Unfortunately, I get the same error and this time, even earlier into my training. This is the suggested reset_episode() I used:
``` 
def _reset(self):
		self.airsim_client.reset()
		self.airsim_client.enableApiControl(vehicle_name=self.drone_name)
		self.airsim_client.arm(vehicle_name=self.drone_name)
		self.airsim_client.simResetRace()
		time.sleep(1.0)
		self.airsim_client.simStartRace(tier=self.race_tier)
		self.takeoff_with_moveOnSpline(takeoff_height=0.4) 
```
If the error message, as you hypothesised (https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/123#issuecomment-553576224), is coming from drone_2 (when it calls moveOnSplineAsync()), then disabling drone_2 (https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/71#issuecomment-538092295) can prove this hypothesis if the crash stops. Else, I can safely assume the crashes occur during resetting of an episode (in my case) when I call moveOnSplineAsync() in takeoff_with_moveOnSpline(). I tried disabling drone_2 and execute my code; no crashes yet. Will let you know how it proceeds.

madratman, 2019-11-16:
https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/123#issuecomment-554159158