microsoft/AirSim-NeurIPS2019-Drone-Racing #115: How is it possible to do a maneuver (Body Rate control) while using "moveOnSplineVelConstraintsAsync()"

uscodorbay, 2019-11-07:
Dear all Guys,

there is question for me how is it possible to do a maneuver movement while drone following the track. I used some Body-Rate movement commands Like: **self.airsim_client.moveByRollPitchYawThrottleAsync(roll = 0 , pitch = 0.4, yaw = 0, throttle = 1, duration = .05).join()**

after **moveOnSplineVelConstraintsAsync()** command and it was not good!! and drone lost it's previous movement
[image]

madratman, 2019-11-07:
It is not possible. 
You can't overlap those two APIs in that manner - in that you can't finetune the behaviour of moveOnSpline's controller that way. You can tune the gains if you want - ctrl+f "gains(" on the API page https://microsoft.github.io/AirSim-NeurIPS2019-Drone-Racing/api.html

we've low level, mid level, and high level control apis. 
the way to go about it would be to write your own agile controller using the low level APIs. 
or tune the behaviour of moveonspline via the controller gains.

uscodorbay, 2019-11-07:
Thank you in advance @madratman. 
yes exactly I need agile drone.  also @madratman Long Path planing is not relate to my Agile Controller, is it right?! 
Can you please tell me how I can make a agile controller and how I have to deal with Long Path Planing!? (Like a predefined big Circle or Paths in your **Game of Drones Racing** Environments )

madratman, 2019-11-07:
Hey, sure. 
I listed some papers in this thread here : https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/44#issuecomment-527738944 Hopefully that helps.

madratman, 2019-11-07:
If you're asking implicitly for an API which can you give a trajectory, which you'll track with your own controller - I think the answer is going to be no, unfortunately. 
In general, we did not do this as this is a competition. moveOnSpline is an intentional black box, so it doesn't become too easy. 
Also, it would be unfair to have this released at this point in time, as other teams have been working on their own trajectory planners and controllers for some time now.  

The papers I point to in the above comment do have open source implementations - a google search might yield them. And, there will be even more I haven't listed for starter code.

closed 2019-11-07 by madratman