microsoft/AirSim-NeurIPS2019-Drone-Racing #116: Engine crash when trying to run baseline_racer_gtp.py

surfii3z, 2019-11-08:
Hello, I try running baseline_racer_gtp.py, following https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing#sample-code.

However I got the engine crash with the following log

> [2019.11.08-06.18.02:468][  0]LogUMG: Display: Widget Class BP_SimHUDWidget_C - Loaded Fast Template.
[2019.11.08-06.18.02:469][  0]LogEngine: Onscreen warnings/messages are now DISABLED
[2019.11.08-06.18.02:469][  0]LogUMG: Display: Widget Class OptionsMenu_C - Loaded Fast Template.
[2019.11.08-06.18.02:699][  0]LogModelComponent: Model /Engine/Maps/Entry.Entry:PersistentLevel.ModelComponent_4 has elements that reference missing vertices. MaxVertex=3, NumVertices=0
[2019.11.08-06.18.02:699][  0]LogModelComponent: Model /Engine/Maps/Entry.Entry:PersistentLevel.ModelComponent_5 has elements that reference missing vertices. MaxVertex=7, NumVertices=0
[2019.11.08-06.18.02:699][  0]LogModelComponent: Model /Engine/Maps/Entry.Entry:PersistentLevel.ModelComponent_5 has elements that reference missing vertices. MaxVertex=11, NumVertices=0
[2019.11.08-06.18.02:699][  0]LogModelComponent: Model /Engine/Maps/Entry.Entry:PersistentLevel.ModelComponent_5 has elements that reference missing vertices. MaxVertex=19, NumVertices=0
[2019.11.08-06.18.02:699][  0]LogModelComponent: Model /Engine/Maps/Entry.Entry:PersistentLevel.ModelComponent_6 has elements that reference missing vertices. MaxVertex=15, NumVertices=0
[2019.11.08-06.18.02:699][  0]LogModelComponent: Model /Engine/Maps/Entry.Entry:PersistentLevel.ModelComponent_7 has elements that reference missing vertices. MaxVertex=23, NumVertices=0
[2019.11.08-06.18.03:106][  0]LogPakFile: New pak file ../../../AirSimExe/Content/Paks/Building99.pak added to pak precacher.
[2019.11.08-06.18.03:125][  0]LogPakFile: New pak file ../../../AirSimExe/Content/Paks/ZhangJiaJie.pak added to pak precacher.
[2019.11.08-06.18.03:274][  0]LogMaterial: Can't compile Mat1_master_unique with cooked content, will use default material instead
[2019.11.08-06.18.03:274][  0]LogMaterial: Warning: /Building99/Building_99/BuildingMeshes/Materials/Mat1_master_unique.Mat1_master_unique: Failed to compile Material for platform GLSL_430, Default Material will be used in game.
[2019.11.08-06.18.03:786][  0]LogUMG: Display: Widget Class BP_LevelLoadButton_C - Loaded Fast Template.
[2019.11.08-06.18.04:305][  0]LogUMG: Display: Widget Class WeatherBoolWidget_C - Loaded Fast Template.
[2019.11.08-06.18.04:305][  0]LogUMG: Display: Widget Class WeatherParamScalarWidget_C - Loaded Fast Template.
[2019.11.08-06.18.04:307][  0]LogSlate: Took 0.001886 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Regular.ttf' (155K)
[2019.11.08-06.18.04:309][  0]LogUMG: Display: Widget Class WeatherParamWindWidget_C - Loaded Fast Template.
[2019.11.08-06.18.04:310][  0]LogTemp: Warning: WidgetBlueprintGeneratedClass Warning, WeatherAPI
[2019.11.08-06.18.04:310][  0]LogTemp: Warning: WidgetBlueprintGeneratedClass Warning, WeatherAPI
[2019.11.08-06.18.04:310][  0]LogLoad: Took 2.003845 seconds to LoadMap(/Game/MainMenu)
[2019.11.08-06.18.04:312][  0]LogSlate: Took 0.001830 seconds to synchronously load lazily loaded font '../../../Engine/Content/EngineFonts/Faces/RobotoBold.ufont' (160K)
[2019.11.08-06.18.04:401][  0]LogPackageName: SearchForPackageOnDisk took   0.060s, but failed to resolve Qualifier_Tier_1.
[2019.11.08-06.18.04:428][  0]LogStreaming: Display: ULevelStreaming::RequestLevel(/Game/FlyingCPP/Maps/Blocks_LevelInstance_1) is flushing async loading
Checking Pak Config[2019.11.08-06.18.04:494][  0]LogLoad: (Engine Initialization) Total time: 4.09 seconds
[2019.11.08-06.18.04:575][  0]LogRenderer: Reallocating scene render targets to support 1600x900 Format 10 NumSamples 1 (Frame:1).
[2019.11.08-06.18.04:579][  0]LogRHI: Using OpenGL program LRU cache: 0
[2019.11.08-06.18.04:668][  0]LogInit: Adding external input plugin.
[2019.11.08-06.18.04:676][  0]LogRenderer: Reallocating scene render targets to support 128x128 Format 10 NumSamples 1 (Frame:1).
[2019.11.08-06.18.04:676][  0]LogContentStreaming: Texture pool size now 1000 MB
[2019.11.08-06.18.04:773][  0]LogRenderer: Reallocating scene render targets to support 1600x900 Format 10 NumSamples 1 (Frame:2).
[2019.11.08-06.18.04:822][  0]LogLinux: Setting swap interval to 'Immediate'
[2019.11.08-06.18.04:828][  0]LogSlate: Slate font cache was flushed
[2019.11.08-06.18.04:834][  1]LogSlate: Took 0.002115 seconds to synchronously load lazily loaded font '../../../Engine/Content/EngineFonts/Faces/RobotoBold.ufont' (160K)
[2019.11.08-06.18.05:411][ 38]LogTemp: spawning player
[2019.11.08-06.18.05:426][ 38]LogLevel: WORLD TRANSLATION BEGIN {0, 0, 0} -> {5690, -100, 202}
[2019.11.08-06.18.05:426][ 38]LogLevel: WORLD TRANSLATION END {5690, -100, 202} took 0.1900 ms
WARNING: Logging before InitGoogleLogging() is written to STDERR
F1108 15:18:11.166389 21817 vertex.cpp:257] Check failed: vertices.size() >= 2 (1 vs. 2) 
*** Check failure stack trace: ***
Signal 6 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=6
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=107920 LargeMemoryPoolOffset=239056 
Failed to find symbol file, expected location:
"/home/jy/AirSim_Training/AirSimExe/Binaries/Linux/AirSimExe.sym"
Malloc Size=81716 LargeMemoryPoolOffset=320788 
Engine crash handling finished; re-raising signal 6 for the default handler. Good bye.
Aborted (core dumped)

What is the cause of this?

madratman, 2019-11-08:
@ericcristofalo can you look at this?
This will happen if moveOnSpline is called with no waypoint in the argument. I am not sure if when you're truncating the waypoints, if there's an assertion being made somewhere.

madratman, 2019-11-08:
hmm, so what is happening is that if waypoint argument of moveOnSpline has only one element, and `add_position_constraint` is set to False, I don't insert the current odometry, thereby leading to this error. 
When the GTP code is sending only a single waypoint, then `add_position_constraint` should be set to True

surfii3z, 2019-11-09:
@madratman Thank you very much for the information. I thought this was the simulator problem.

madratman, 2019-11-12:
https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/pull/124 fixes this

closed 2019-11-12 by madratman